alexandrebarachant/muse-lsl #1: Step-by-step process to connect with Muse

reallyFrancisco, 2017-02-07:
Hello, 
I'm a new user to Muse and LSL and curious how I can get connected to a Muse device with LSL. I've currently downloaded and installed all dependencies on Mac OSX (10.11.6). However, when I try running muse-lsl.py in terminal I get: "The command was not found or was not executable: gatttool."

I understand that gatttool works on linux but how can I get recording data using my current set up?

Thanks in advance.

alexandrebarachant, 2017-02-07:
Hi,
the BLE interfacing rely on the python package `pygatt`, that require blueZ / Gatttool for linux or a dedicated [BLE dongle](http://www.silabs.com/products/wireless/bluetooth/bluetooth-smart-modules/Pages/bled112-bluetooth-smart-dongle.aspx) for windows (and apparently on OS X too, see this issue https://github.com/peplin/pygatt/issues/67)

I have hard time to find a true cross-platform BLE library. I'm working on ubuntu, but I also happens to have a BLED112 dongle to test it on windows, but I don't have access to a Mac.

I'm afraid i don't have an immediate solution for you. If you can find a BLE python library that works on OSX, let me know.

Otherwise, you can try running it on a virtual machine with ubuntu.

andira14, 2017-05-16:
I have a problem, I ran this on ubuntu mate and when I typed python muse-lsl.py --address 00:55:DA:B0:40:50

it shows error like this: 
`
WARNING: No module named pexpect
Traceback (most recent call last):
  File "muse-lsl.py", line 3, in <module>
    from pylsl import StreamInfo, StreamOutlet, local_clock
  File "/usr/local/lib/python2.7/dist-packages/pylsl/__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "/usr/local/lib/python2.7/dist-packages/pylsl/pylsl.py", line 1147, in <module>
    lib = CDLL(libpath)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 362, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/local/lib/python2.7/dist-packages/pylsl/liblsl32.so: cannot open shared object file: No such file or directory`

How can I solve this? I have no idea what the problem might be since I'm not good in linux configuration and haven't program using python. thank you.

alexandrebarachant, 2017-05-16:
This is a problem with pylsl. 
Try to uninstall and reinstall the package : 

to uninstall (try multiple time if you need to) `pip uninstall pylsl` 
then to install `pip install pylsl`

you might need a `sudo` if there is problem with admin right

andira14, 2017-05-16:
@alexandrebarachant 

Still the same result. I tried using `python3` to run the program and also install using `pip3` but still the same result.

is it compatible with raspbian? because I'm using ubuntu on a raspberry pi

alexandrebarachant, 2017-05-16:
I never tried with a Raspberry, but I know that @jona-sassenhagen managed to make it works.

from your error, it looks like you need to install pexpect

andira14, 2017-05-16:
@alexandrebarachant I cannot find muse project in his github. Thank you for your help sir :)

alexandrebarachant, 2017-05-16:
he might pass by here and answer your question :)

jona-sassenhagen, 2017-05-16:
The pylsl error is because the liblsl.so file was compiled for intel. (At least that was the error I got.) you can recompile the file from the sources. You will have to change the sources a bit for that though, there is an issue on the lsl github discussion that. I dont have my pi with me right now though, can't give a detailed solution today.

andira14, 2017-05-16:
@jona-sassenhagen okay, thank you for your help. Maybe I'll wait for your solution and try other references while waiting.

jona-sassenhagen, 2017-05-16:
@andira14 I'm still not near my Pi, but I'm at my computer now.

IIRC, essentially, you have to

* check if your problem is indeed that you have an intel compiled file by doing e.g. `file  /usr/local/lib/python2.7/dist-packages/pylsl/liblsl32.so`
* go here: ftp://sccn.ucsd.edu/pub/software/LSL/ and download the sources (I picked the most recent file - liblsl-C-C++-1.12.zip IIRC)
* compile them *on your Pi*:
* fix the sources by removing a legacy call, see here: https://github.com/sccn/labstreaminglayer/issues/74
* otherwise follow the instructions in the README
* replace the file  /usr/local/lib/python2.7/dist-packages/pylsl/liblsl32.so with the resulting .so file 

Now importing pylsl should work and `OSError: /usr/local/lib/python2.7/dist-packages/pylsl/liblsl32.so: cannot open shared object file: No such file or directory` should be gone.

jona-sassenhagen, 2017-05-17:
@andira14 did you try it?

andira14, 2017-05-18:
@jona-sassenhagen I'm sorry for my late reply. I haven't tried it yet because I have other urgent work to be done first. I'll try it maybe next week and I'll let you know the result. Thank you for your reference.

alex-kachur-jibo, 2017-10-16:
Experiencing the same problem here. 
@jona-sassenhagen may I ask you to upload somewhere pre-built lsl library for raspberry pi?

olka, 2017-10-16:
Experiencing the same problem here.
@jona-sassenhagen may I ask you to upload somewhere pre-built lsl library for raspberry pi?

alex-kachur-jibo, 2017-10-16:
I've compiled it but getting following error:

python muse-lsl.py --name Muse-3123
Found device Muse-3123 : 00:55:DA:B3:31:23
Traceback (most recent call last):
  File "muse-lsl.py", line 44, in <module>
    muse.connect()
  File "/home/pi/Projects/muse-lsl/muse/muse.py", line 56, in connect
    self._subscribe_eeg()
  File "/home/pi/Projects/muse-lsl/muse/muse.py", line 101, in _subscribe_eeg
    callback=self._handle_eeg)
  File "/home/pi/.local/lib/python2.7/site-packages/pygatt/device.py", line 144, in subscribe
    self._notification_handles(uuid)
  File "/home/pi/.local/lib/python2.7/site-packages/pygatt/device.py", line 120, in _notification_handles
    value_handle = self.get_handle(uuid)
  File "/home/pi/.local/lib/python2.7/site-packages/pygatt/device.py", line 208, in get_handle
    raise exceptions.BLEError(message)
pygatt.exceptions.BLEError: No characteristic found matching 273e0003-4c4d-454d-96be-f03bac821358

olka, 2017-10-16:
So the reason is this issue with pygatt:
https://github.com/peplin/pygatt/issues/152
Do not use latest version until this one is fixed!
`pip install pygatt==3.1.1` works well though

them4ra, 2017-10-17:
I'm getting this kind of error

No handlers could be found for logger "pygatt.backends.gatttool.gatttool"
Traceback (most recent call last):
  File "muse-lsl.py", line 44, in <module>
    muse.connect()
  File "/home/pi/Desktop/ERP/muse-lsl/muse/muse.py", line 47, in connect
    address = self.find_muse_address(self.name)
  File "/home/pi/Desktop/ERP/muse-lsl/muse/muse.py", line 68, in find_muse_address
    list_devices = self.adapter.scan(timeout=10.5)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 322, in scan
    raise BLEError(message)
pygatt.exceptions.BLEError: BLE adapter requires reset after a scan as root- call adapter.reset()

them4ra, 2017-10-17:
I'm getting an error like this:

`No handlers could be found for logger "pygatt.backends.gatttool.gatttool"
Traceback (most recent call last):
  File "muse-lsl.py", line 44, in <module>
    muse.connect()
  File "/home/pi/Desktop/ERP/muse-lsl/muse/muse.py", line 47, in connect
    address = self.find_muse_address(self.name)
  File "/home/pi/Desktop/ERP/muse-lsl/muse/muse.py", line 68, in find_muse_address
    list_devices = self.adapter.scan(timeout=10.5)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 322, in scan
    raise BLEError(message)
pygatt.exceptions.BLEError: Unexpected error when scanning: Set scan parameters failed: Operation not permitted
`

alexandrebarachant, 2017-10-17:
@them4ra try the fix in the last message from this : https://github.com/alexandrebarachant/muse-lsl/issues/14

them4ra, 2017-10-17:
Hi @alexandrebarachant . thanks for such a quick reply!

I got that worked out now thanks so much- I was a little too flustered to even notice that this was already handled in another thread. Alas, now I've run into another hitch. When I run muse-lsl.py I'm getting an error:

```
No handlers could be found for logger "pygatt.backends.gatttool.gatttool"
Traceback (most recent call last):
  File "muse-lsl.py", line 44, in <module>
    muse.connect()
  File "/home/pi/Projects/muse-lsl/muse/muse.py", line 47, in connect
    address = self.find_muse_address(self.name)
  File "/home/pi/Projects/muse-lsl/muse/muse.py", line 68, in find_muse_address
    list_devices = self.adapter.scan(timeout=10.5)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 322, in scan
    raise BLEError(message)
pygatt.exceptions.BLEError: BLE adapter requires reset after a scan as root- call adapter.reset()

```
Sorry if I'm bombarding you, been trying to get this working for a few days and I'm starting to feel the desperation sink in.

alexandrebarachant, 2017-10-17:
Hi,

I think you have to run the code as root.

All those bugs are related to pygatt. I only have a limited knowledge about it and you should try to open a ticket directly to them.

Saying that, you can bypass the scan (which seems to cause trouble) by providing the MAC address of your muse headset.

them4ra, 2017-10-17:
OK, thanks for your help, I'll do that and start a ticket with pygatt. 
Running muse.lbl.py with the MAC address just times out so it's probably  to do with pygatt doing something weird.

```
No handlers could be found for logger "pygatt.backends.gatttool.gatttool"
Traceback (most recent call last):
  File "muse-lsl.py", line 44, in <module>
    muse.connect()
  File "/home/pi/Projects/muse-lsl/muse/muse.py", line 52, in connect
    self.device = self.adapter.connect(self.address)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 384, in connect
    raise NotConnectedError(message)
pygatt.exceptions.NotConnectedError: Timed out connecting to 15:36:BA:23:C8:EE after 5.0 seconds.
```
Thanks for your help. I can't wait to get to try this out, this is a really cool project you developed. You have a lot to be proud of.

alexandrebarachant, 2017-10-17:
Thanks.

from time to time, you have to run the command several time before it start working. Its for sure not as stable as it should be, but there is no other option than using pygatt with python.

them4ra, 2017-10-22:
I got muse-lsl to work on my Raspberry Pi 3. Thanks so much for everyone's help! I thought I'd help out others that might be trying to do the same thing so here's a list of everything that I needed to do. You can find all this info in different posts from different issues so having them all in one place might be helpful.

1. Install Pygatt version 3.1.1, the newest version (3.2) will no work as of writing this.

`pip install pygatt==3.1.1 `

2. As @jona-sassenhagen  said, If you're using Raspberry Pi your going to need to replace liblsl32.so with one that has been compiled for ARM architecture. I was able to find it already compiled [here](https://github.com/chkothe/pylsl/pull/2) Remove the old file and replace it with the one you get. (Thanks so much @micuat !) 

3. As @alexandrebarachant said and found in the documentation, you have to run this thing as root so run the following line to set it up so it gets run as root every time: 

`sudo setcap 'cap_net_raw,cap_net_admin+eip' `which hcitool``

4. Run muse-lsl with the MAC address of the Muse headband. You can get the MAC address by running 

`sudo hcitool -i hci0 lescan`

and then turning on the headband as the LE scan looks for it. It'll read out the MAC address once it's found. 

5. Finally run muse-lsl with the MAC address.

`python muse-lsl.py --address [MAC ADDRESS HERE]`

Then it's hopefully finding your Muse and sending EEG data! I had to still install the dependencies that were needed for lsl-viewer (matplotlib, scipy, pandas, etc.) and then it all started working out. I used a list I found [here](http://wyolum.com/numpyscipymatplotlib-on-raspberry-pi/) to help install most of the necessary tools.

Again, I want to thank @alexandrebarachant  for sharing this project and helping out! This is a really awesome project and I am really happy that I can have a chance to work with it.

francescacoo, 2017-12-29:
Hi everyone, 
not sure if I should post here or open a new request.
I am trying to run the code on Ubuntu and I am suddenly facing the same problem (pygatt.exceptions.NotConnectedError:)

I already tried to troubleshoot but cannot solve.. and the strange thing is that everything was working before on the same Ubuntu machine, same dongle, same Muse.

Anyway, I can connect to the muse with bluetoothd, it pairs and it connects.
When I run the python muse-lsl --address MAC I get:

`No handlers could be found for logger "pygatt.backends.gatttool.gatttool"
Traceback (most recent call last):
  File "muse-lsl.py", line 41, in <module>
    muse.connect()
  File "/media/dev/muse/muse-lsl-master/muse/muse.py", line 43, in connect
    self.device = self.adapter.connect(self.address)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 384, in connect
    raise NotConnectedError(message)
pygatt.exceptions.NotConnectedError: Timed out connecting to 00:55:DA:B0:05:B1 after 5.0 seconds.
`
any idea on how I could troubleshoot this?
I already tried to uninstall and reinstall dependencies and bluetooth manager..

thank you!

ljy1379, 2018-01-24:
@them4ra Good job. Thank you!

hoikc, 2018-02-01:
Hello everyone!
I followed method of @them4ra and I successed to excute muse-lsl in my pi.
However i got problem of 'missing sample'.
I just get sample using StreamInlet as follows.

if len(streams)==0:
    raise(RuntimeError("Can't find EEG stream!"))
inlet = StreamInlet(streams[0],max_chunklen=buf)

while 1:
    try:
        sample, timestamp = inlet.pull_chunk(timeout=1.0, max_samples=12)

However after few minutes, cpu usage of pi suddenly goes up and 'missing sample' occurs.
Is there anyone who have this problem in pi?

sylvchev, 2018-04-13:
No such problem on Pi 3 with Raspbian. Bought a BLED112 dongle but the RPi 3 is handling bluetooth natively and connect well with Muse. 
I had one strange issue, with Muse cap blinking with a pattern different from the usual one. The muse-lsl.py script throw the same error than the pygatt-3.2.0 (even if the pygatt-3.1.1 is installed):
`pygatt.exceptions.BLEError: No characteristic found matching 273e0003-4c4d-454d-96be-f03bac821358`
To solve the error, I had to connect once the headband with the official phone app, then it worked again.

alexandrebarachant, 2018-04-13:
@sylvchev Weird. perhaps an issue with updating the firmware of the band ?

sylvchev, 2018-04-14:
No, unfortunately. It happens each time after a full charge of the headband battery. I suspect that the band is reset in some specific mode, waiting a command to switch to normal mode. I will try to snoop the BLE exchange.

jdpigeon, 2018-06-01:
^ I've just replicated this bug. Super weird.

1. Using muselsl to connect and stream data on Ubuntu 18 with no problems. Using pygatt 3.1.1 and having already run the setcap command.
2. Charge a newly-obtained 2016 headband from dead all the way to full charge
3. Attempting to connect to headband and the indicator does a 'broken half blink' and then the `pygatt.exceptions.BLEError: No characteristic found matching 273e0003-4c4d-454d-96be-f03bac821358` error is returned.
4. Connect the Muse to the Calm app on Android. Interestingly, this takes longer than usual and half way through the headband does the broken blink again before connecting successfully.

This has got to be something in the Muse firmware. The device starts in a special BLE mode that only the official app can handle?

jdpigeon, 2018-06-01:
Corrected in https://github.com/alexandrebarachant/muse-lsl/pull/37

shenjianaixuexi, 2018-12-10:
@andira14  have  you solved the os error？ can U give me some advice？ thank U

shenjianaixuexi, 2018-12-10:
@jona-sassenhagen . hello ,you solution address is gone .can you give it to me ? 457o7482@qq.com
i have same question with os error above mentioned  above

jdpigeon, 2018-12-12:
Hey @shenjianaixuexi , in addition to uninstalling and reinstalling `pylsl`, I also recommend having a look at this: https://github.com/pwnsauce8/ZSWI/wiki/using-pyLSL-module-on-raspbian

mauroibarra, 2019-09-07:
cordial saluda, 
estoy intentando trabajar con el muse desde mi Rasberry pi 3 B+
tengo el siguiente problema:

Traceback (most recent call last):
  File "/usr/local/bin/muselsl", line 6, in <module>
    from muselsl.__main__ import main
  File "/usr/local/lib/python3.7/dist-packages/muselsl/__init__.py", line 1, in <module>
    from .stream import stream, list_muses
  File "/usr/local/lib/python3.7/dist-packages/muselsl/stream.py", line 2, in <module>
    from pylsl import StreamInfo, StreamOutlet
  File "/usr/local/lib/python3.7/dist-packages/pylsl/__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "/usr/local/lib/python3.7/dist-packages/pylsl/pylsl.py", line 1147, in <module>
    lib = CDLL(libpath)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/local/lib/python3.7/dist-packages/pylsl/liblsl32.so: cannot open shared object file: No such file or directory

les cuento que con: sudo pip install muselsl, solo funciono con pip3.

he seguido los procedimientos encontrados en https://github.com/alexandrebarachant/muse-lsl, y ya he tenido en cuenta muchos de los comentarios de este repositorio.

estaré muy agradecido con su ayuda.

mauroibarra, 2019-09-07:
cordial
greeting,
I am trying to work with the museum from my Rasberry pi 3 B +
I have the following problem:

pi@raspberrypi:~ $ muselsl stream
Traceback (most recent call last):
  File "/usr/local/bin/muselsl", line 6, in <module>
    from muselsl.__main__ import main
  File "/usr/local/lib/python3.7/dist-packages/muselsl/__init__.py", line 1, in <module>
    from .stream import stream, list_muses
  File "/usr/local/lib/python3.7/dist-packages/muselsl/stream.py", line 2, in <module>
    from pylsl import StreamInfo, StreamOutlet
  File "/usr/local/lib/python3.7/dist-packages/pylsl/__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "/usr/local/lib/python3.7/dist-packages/pylsl/pylsl.py", line 1147, in <module>
    lib = CDLL(libpath)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/local/lib/python3.7/dist-packages/pylsl/liblsl32.so: cannot open shared object file: No such file or directory

He followed the procedures found at https://github.com/alexandrebarachant/muse-lsl, and already took into account many of the comments in this repository.

I will be very grateful with your help.

jdpigeon, 2019-10-22:
@mauroibarra, my best recommendation is to follow the guide [here](https://github.com/alexandrebarachant/muse-lsl#linux) for getting PyLSL working on Rasbian. the error you've posted looks like you have a faulty pylsl installation.

I'm going to close this issue because it's old and has gone offtrack a few times

closed 2019-10-22 by jdpigeon