alexandrebarachant/muse-lsl #2: Bluetooth use on Windows 7

Cataboum, 2017-04-11:
Hello,
I'm trying to use Muse and LSL, but I'm on Windows 7. 
I installed a BLE112 dongle with the driver. But when I try to launch the `python muse-lsl.py --adress myaddress`
I have this type of issue : `No BGAPI compatible device detected`

I wanted to know if it's because it didn't find the BLE112 dongle or the Muse headband. 

Thanks

alexandrebarachant, 2017-04-12:
Did it solve your problem ? If so, please close the issue.

Cataboum, 2017-04-12:
It doesn't really solve my problem, but I think that the problem it's between my computer and the dongle. The error that I have is :  `"Cannot configure port, some setting was wrong"` 
I use the good port, I can detect it with the Serial library. But they are just the BLED112 drive installed on my computer, but no bluetooth module installed. I don't know if it's necessary to use your program.

I'm going to try on another computer which already have bluetooth.
I'll come to you if I have other problems, or if I find a solution for my main computer.

Thanks

alexandrebarachant, 2017-04-12:
Okay. 
you should look at pygatt, there was an issue with a very similar problem : 

https://github.com/peplin/pygatt/issues/100

Cataboum, 2017-04-12:
This issue  didn't help me because my bluetooth port is 6.

I wanted to know, before using your program, did you paired your Muse headband ? Because me I cannot, and it seems that the BLED112 dongle don't need to pair, but when I try to use it with my Muse, I put my Muse in pairing mode, and it's not working.

alexandrebarachant, 2017-04-12:
Pairing is not necessary. Actually, you should not be able to pair, since the dongle embedd the full BLE stack.

If you install the SDK for the bled112 dongle, you should be able to use their program and try to connect to the muse directly.

Cataboum, 2017-04-13:
I downloaded the SDK to use with the BLED112 dongle, but when I launch it and I scan, it didn't find my muse, but it's able to find others bluetooth.

alexandrebarachant, 2017-04-13:
Is your headset the 2016 muse headset ?

Cataboum, 2017-04-13:
Mmmmh, no it's a 2014 muse headset.

alexandrebarachant, 2017-04-13:
Okay. 2014 headset is Bluetooth and not BLE. it wont work with this code nor with the BLED112 dongle.

This code is only compatible with the 2016 BLE headset. As a matter of fact, the reason of the existence of this code is because Muse-io (interaxeon way to get raw data) doesn't support the 2016 headset.
If you have a 2014 headset, then you can use their code to get raw data.

I guess i have to put a bigger warning in the Readme

Cataboum, 2017-04-13:
Ah ok, so it's not working because of the different bluetooth protocol !? The only problem is to use it with my computer which doesn't have bluetooth. So I need another type of dongle, compatible with the 2014 muse headset.

Thank you for your answers :)

closed 2017-04-13 by Cataboum

alexandrebarachant, 2017-04-13:
Yes, it's a different Bluetooth protocol, and the data format is also a bit different. So even with a compatible dongle, you will not be able to use this code.

A regular Bluetooth dongle and muse's SDK will do the job just fine.

Cataboum, 2017-04-13:
Ok, I'm going to buy another dongle for my computer.
Thank you !