alexandrebarachant/muse-lsl #3: Is BLED112 dongle problem?

fromnear, 2017-04-20:
[image]

1. At the first I got "pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected" message.
After search, I changed the code row 14, None to "COM1"

2. then I could get different message, "pygatt.backends.bgapi.exceptions.ExpectedResponseTimeout: Timed out after 1.000000s waiting for [<ResponsePacketType.gap_set_mode: 58>"

I am using Next v10.10 Bluetooth 4.0 usb dongle(Korean company), Python 3.5.2, Windows 10 Pro, Muse 2016. What would be problem?

Q1. Is a bluetooth dongle must be USB dongles made by Bluegiga, BLED112 only?
Does the codes works on other bluetooth low energy 4.0 dongles?

Q2. Do I have to use Bluesoleil things for pairing? or just OK with basic windown bluetooth tool?

It will be very nice if I could run your code. :)

alexandrebarachant, 2017-04-20:
Hi,

1. The first error is because pygatt is looking for a BLED112 dongle, so auto-detection does not work with other dongle. you have to specify the COM port manually.
2. The second error appears during initialization of the dongle. The protocol is probably different / incompatible so the code raise an error when the dongle does not respond.

Q1. I think that's a question to ask to the pygatt authors https://github.com/peplin/pygatt

Q2. You don't need to pair the device at all.

fromnear, 2017-04-20:
Ok, I'll try to ask to pygatt authors. Please check the followings before I close this.

1. Have you tried your code on windows? Because pygatt authors said "GATTToolBackend requires Linux (i.e. not Windows compatible)". And you said "pygatt works on linux and should work on window and OSX provided that you have a BLED112 dongle". I wonder what "should work on windows" supposed to mean.

2. Do you use Blugiga BLED112 dongle? Does it look like this? I want to confirm it before I order one.
[image]

alexandrebarachant, 2017-04-20:
1. Yes, i tried on windows with a BLED112 dongle. The code works (you have to install pygatt from the sources). I will update the README.

2. Yes, this is the dongle.

fromnear, 2017-04-20:
Very kind of you. Thank you for quick answer.
I appreciate that.

I'll try your code with that dongle. Order one over the pacific ocean. Hope it will work.

closed 2017-04-20 by fromnear

alexandrebarachant, 2017-04-20:
The other solution is to boot with ubuntu on a liveUSB :) should works out of the box (once you installed python)