alexandrebarachant/muse-lsl #4: BLEError, no characteristic found matching UUID

pdpino, 2017-04-29:
Hi,
I get the following error when trying to connect to the headband:
`pygatt.exceptions.BLEError: No characteristic found matching 273e0003-4c4d-454d-96be-f03bac821358`
which is raised when trying to subscribe the first eeg channel in the `_subscribe_eeg()` method. The weird part is that until a week ago I used the same code and it worked fine, I was able to connect and receive data from the heaband.

I tried connecting with the headband via `gatttool`, and I saw that there is no uuid matching the one from the error on the list (only one that is almost the same):
[image]
[image]

Although I am new with bluetooth connections, so I'm not sure if this is relevant at all. Any ideas?
Thanks!

alexandrebarachant, 2017-05-01:
Interesting.
Is this error recurrent i.e happens every times ?

i really have no idea where it came from.

pdpino, 2017-05-04:
It happened just the one time.
After about a day of trying everything (power off and on, disconnect bluetooth adapter, pair unpair, block unblock, trust untrust, charging battery, etc) using `hcitool`, `gatttool` and `bluetoothctl`, I don't know how but it got fixed (I don't even know if it was something that I did or not). 
Anyway, thanks again.

alexandrebarachant, 2017-05-04:
Ok thanks for the update.

pygatt scans the uuid before it can subscribes to it. It might have happened that the muse timed out before sending the complete list. 

I'm closing the issue, but this remains a mystery.

closed 2017-05-04 by alexandrebarachant

jdpigeon, 2018-04-15:
Getting this error with one of my headbands. Very mysterious. 

My other headband works, though.

pdpino, 2018-04-15:
Check the pygatt version (see README, common issues: [link](https://github.com/alexandrebarachant/muse-lsl/blob/master/README.md#common-issues)).
Another option is charging the battery.
If neither works I have no clue.

jdpigeon, 2018-04-15:
After turning headband on and off and connecting to it through EEG 101 (unusually long connection time) it's working. Definitely a hardware issue.