alexandrebarachant/muse-lsl #5: lsl-viewer doesn't work over ssh/X forwarding

jona-sassenhagen, 2017-05-16:
My ideal setup would be to ssh over wifi into my raspberry pi. However, `lsl-viewer.py` doesn't work that way - it only shows one frame and then halts. This means I can't check if my channels are connected without buying a monitor for my Pi. Is this something that could in principle be solved, or is updated MPL over ssh/X forwarding not possible?

alexandrebarachant, 2017-05-17:
LSL is accessible through local network.
you can run muse-lsl.py on the raspberry, and run the lsl-viewer on any computer on your local network.

jona-sassenhagen, 2017-05-17:
Ah! That's interesting. I'll see if I can get that set up.

aj-ptw, 2017-10-22:
I've been trying to get this working too. I get error 
```
OSError: /usr/local/lib/python2.7/dist-packages/pylsl/liblsl32.so: cannot open shared object file: No such file or directory
```

I think the program ran properly the first time. But even when the script ran the first time on the PI i could not see a stream with `lsl-viewer.py` and `lsl-viewer-V2.py` failed with a different error.

closed 2019-10-22 by jdpigeon