alexandrebarachant/muse-lsl #6: Data in lsl-viewer flaps up and down at beginning and end

jdpigeon, 2017-06-17:
Just got data visualized in lsl-viewer.py while setting up a Ganglion OpenBCI through the [NeurotechBerkely lab](https://github.com/NeurotechBerkeley/bci-course/tree/master/lab7)

The data looks good *in the middle* of the graph, but it flaps up and down intensely at the beginning and the end. Is this something to do with auto-scaling or auto centering of the plot along the y axis?

[image]

alexandrebarachant, 2017-06-18:
This is transient response of the filtering. You can disable the filtering by hitting the key 'd'. 

I used a very dirty filtering implementation (both inefficient and inelegant). We can improve that with a truly online filtering implementation.

jdpigeon, 2017-06-18:
It feels good to know.

I have some experience with online filtering implementation through EEG 101, maybe I can look into this if lsl-viewer has a life ahead of it.

jdpigeon, 2017-06-19:
The flapping is much worse for OpenBCI than Muse. Why would this be? Don't they have similar ~250ms sampling rates?

Also, what would a better filter implementation look like? Currently it just runs the scipy filtfilt function on all data points every time. Should there be a seperate buffer for filtered samples or some way to keep track of the internal state of the filter over time?

alexandrebarachant, 2017-06-19:
OpenBCI does not have a high pass filter, so there is a stronger baseline and that creates more transient response (filter are initialized with 0 coefs)

> Also, what would a better filter implementation look like? Currently it just runs the scipy filtfilt function on all data points every time. Should there be a seperate buffer for filtered samples or some way to keep track of the internal state of the filter over time?

Exactly. you could use `lfilter`. i think it can return internal state of the filter to be reused

jdpigeon, 2017-06-21:
Sweet. I changed the filter to an `lfilter` that keeps track of filter state with `zi`

Looks good on the Muse. I'll throw a PR once I try it out on the OpenBCI

alexandrebarachant, 2017-06-21:
great !

With lfilter, you will have larger transient at the beginning of the recording, but it should stabilize quickly.

closed 2017-07-11 by alexandrebarachant