alexandrebarachant/muse-lsl #10: Connecting MUSE via BLED dongle?

CeciaZ, 2017-06-27:
Hello,
I just started to use Muse and want to connect MUSE 2016 to my Windows 7 computer with LSL. But my MUSE couldn't be connected to my computer via Bluetooth and thus I couldn't get the MAC address of my MUSE. Could you please tell me how to get the MAC address in such situation? I saw that the pygatt only works on Windows with BLED112 dongle. As I didn't use a BLED dongle, does the problem lie in lack of dongle?
Thank you very much in advance!

alexandrebarachant, 2017-06-27:
Yes, on windows, you need a BLED112 dongle.

on linux, it works out of the box. 

you might want to try this muse lsl (javascript) https://github.com/urish/muse-lsl, it might work without a dongle

CeciaZ, 2017-06-28:
Hi Alexandre,
thank you very much for your quick and helpful answer! I'll try it out. :)

CeciaZ, 2017-07-10:
Hello,
I still couldn't get the MAC address of my MUSE headband and thus it's impossible to stream data. I inquired the MUSE customer services and they told me that MUSE 2016 is not compatible with computers. Could you please tell me, how did you get the MAC address when the MUSE isn't connected to the computer? (Btw, I have a BLED112 dongle now.)
Thank you very much!

alexandrebarachant, 2017-07-11:
MAC address is unique to the device, you can get it with a smartphone (there is many apps for that).
As soon as i have time, i will implement an auto-discover function for the toolbox, so we don't need to find the mac address.

in the meantime, i added a script `scan_devices.py` that will help you to get the mac address.

alexandrebarachant, 2017-07-11:
Okay, i actually made the auto detection, so you can update the repository and just run the script with no address

CeciaZ, 2017-07-24:
Hi Alexandre!

Thank you very much for your reply! But I still have some problem... I tried both possibilities (auto detection and with MAC address), but it still doesn't work. It always gives me the error feedback "No BGAPI compatible device detected". I did change the COM port as stated by peplin but it didn't work... Could you please help me again? Thank you so much!

alexandrebarachant, 2017-07-24:
okay,  This is because muse-lsl does not detect your dongle.
lets try to tell muse-lsl which COM port to use.

1. find the COM port of your BLED112 dongle (let say it is COM9)
2. `python muse-lsl.py --interface COM9`

CeciaZ, 2017-07-25:
Thank you!
I tried that, but it still didn't work. Still said "No BGAPI compatible device detected"... Do you maybe know why?

alexandrebarachant, 2017-07-25:
there is a problem with the detection of your dongle.
I only have a limited experience with windows and the BLED112 so your best chance is too look at pygatt and open a issue if you can't find a solution.

this might be useful : https://github.com/peplin/pygatt/issues/118

CeciaZ, 2017-07-31:
Thank you so much! I finally worked out the problem. It's very helpful of you!

closed 2017-07-31 by alexandrebarachant

kylemath, 2017-09-05:
how did you solve this problem @CeciaZ ?

amisepa, 2022-01-27:
Hi @CeciaZ and @alexandrebarachant 

I have the same issue. How did you solve it? 

My Muse is connected to my BLED112 dongle succesfully through the Bluegiga GUI, but Bluemuse does not detect it. 
I tried the command lines from this post (https://github.com/alexandrebarachant/muse-lsl) to connect to the Muse name or to the port. 
e.g.,:  `muselsl stream --interface COM5`
but it always just opens Bluemuse and nothing happens, or I get an error. 
[image]

This is with the muse-lsl from the link above. Should I try with this muse-lsl instead: https://github.com/urish/muse-lsl

Any other suggestion?

I would really appreciate any help, thanks!