alexandrebarachant/muse-lsl #11: EEG raw data from muse

shirans, 2017-07-29:
First of all, I’ve been playing with the code in this repo and it’s very helpful, thanks!

I managed to reproduce the p300, but I haven't figure out the scale of the values I see.  The general saying is that eeg data is around  10-100 uv. According to the docs in Muse, values on /muse/eeg are  0-1600 uv. 

I also tried to reproduce using data from muse monitor. I’m attaching a description of the two data frames - the df from this repo (sub1 session1) and the df I created using muse monitor data.
(df = raw.to_data_frame(), raw is the same raw from the notebook, before filtering)

- How come I see negative values?
- There is a major difference in values range between the data from muse monitor ( 600 - 1100 uv) vs directly from the headband (-300 to 250). Any idea what the difference mean? 
- Should I ignore anything out of the 10-100 uv range (more or less)? Is that the meaning of the reject parameter of MNE.Epochs ? 

[image]

alexandrebarachant, 2017-07-29:
Hi,
I have no experience with muse monitor, so i will not be very helpful.
All i can say is the following : 
- my code save data in uV
- This is a reverse engineering of the BLE data from the muse. Data are streamed as 12 bits integer, that i then convert to float here : https://github.com/alexandrebarachant/muse-lsl/blob/master/muse/muse.py#L113. I center the data (because the hardware is AC coupled, so the signal should be centered around 0). it seems that muse monitor does not center data, but as you can see, the mean is around 800uV (mid range of the 0-1600uV). I also scale the data to span the 12 bits on the 2mV input range specified in the hardware spec of the device : http://developer.choosemuse.com/hardware-firmware/hardware-specifications. This scaling can be wrong.
- Large deviation from the center (after high pass filtering or baseline correction, >200uV) are generally due to artifact. a common practice is to reject epochs that contains artifacts.

pdpino, 2017-08-02:
Hi,
I was wondering why those numbers in https://github.com/alexandrebarachant/muse-lsl/blob/master/muse/muse.py#L113, thanks. 
I still have one question though, are you sure the eeg data is encoded as unsigned integers?

alexandrebarachant, 2017-08-03:
2048 is the center of the 12 bits (4096).
0.48828125 is because the datasheet say 2mv p-p input range (2000uV/4096)

pdpino, 2017-08-03:
Sorry, I misexplained myself, I already got that from your latter comment. 
My question is why unsigned ints? couldn't the muse protocol use signed integers or floats? Floats of 12bit would be weird, and if the number is positive it makes sense to be unsigned, but are you sure of it?

alexandrebarachant, 2017-08-03:
oh sorry about that :)

yes, i'm pretty sure (p-value < 0.01) it is unsigned, at least with the headset i have in hand.
it is difficult to misread a signed int for an unsigned, because when the signal cross zeros, the signed bit flip and if you read it as an unsigned, it create large jump in the data.

pdpino, 2017-08-03:
Ok, great.
@shirans I had similar doubts about the units, I guess this discards that is a problem with the data delivered by muse-lsl. Maybe trying without centering the data around 0?

alexandrebarachant, 2017-08-03:
To be clear, the units are a little bit arbitrary, but it's roughly in uV. As you pointed out, muse monitor values are from 0 to 1600uV, then my 2mV are off.

data are already centered around the mid scale of the ADC. there might be a little offset due to offset error in the hardware signal amplification chain, but since it is AC coupled, it should be around zeros. The muse monitor app return unsigned values, so there is an error on their side.

But in any case, centering does not matter at all, because you always high pass the data, or apply baseline correction, effectively removing the offset.

If for some reason your application need absolute precision, then i would suggest to contact muse and ask them for the equation to convert adc values into voltage :)

shirans, 2018-01-19:
Sorry it's been a while :)
I just tried to dig into it again, bottom line is I didn't find a better way to parse it. 
Just wanted to note that according to the docs the non scaled range is "0.0 - 1023.0 unitless". It also says "If you use the –no-scale command-line option with MuseIO, you get the EEG values as seen by the analog to digital converter."
http://developer.choosemuse.com/research-tools/available-data#EEG_Data

Can't figure out why the non scaled has a 10 bits range while the data from the device comes in 12 bits range. Maybe cause 2014 version was sending 10 bits?

A naive straightforward way would be to normalize the data to 0-1680 range, i.e.
data = res[1:]
data = 0.48828125 * (np.array(data))
1680 * ((data) - data.min()) / (data.max() - data.min())

But I'm feeling it's wrong doing it on 12 samples batches, and also not sure it really would scale to uv.

closed 2019-10-22 by jdpigeon