alexandrebarachant/muse-lsl #13: Linux: hciconfig is deprecated and needs to be installed via bluez-utils-compat

luzpaz, 2017-09-06:
Heads-up: When running muse-lsl.py it will complain about not finding `hciconfig` 
I did a little googling a learned from https://wiki.archlinux.org/index.php/bluetooth#Troubleshooting that hciconfig is deprecated in favor of `btmgmt (and bluetoothctl?)`
I needed to install hciconfig via https://aur.archlinux.org/packages/bluez-utils-compat which installs deprecated tools as well. 

Running Manjaro Linux 17.03

alexandrebarachant, 2017-09-06:
hi,
this is again an issue for [pygatt](https://github.com/peplin/pygatt).

please report that so they can advise

Alex

alexandrebarachant, 2017-09-06:
actually, i think they already have an issue for that : https://github.com/peplin/pygatt/issues/112

closed 2019-10-22 by jdpigeon