alexandrebarachant/muse-lsl #14: Problems running muse-lsl.py

luzpaz, 2017-09-06:
**Specs:** 
```
System:    Host: beast Kernel: 4.9.47-1-MANJARO x86_64 (64 bit gcc: 7.1.1) Desktop: Xfce 4.12.4 (Gtk 2.24.31)
           Distro: Manjaro Linux
Machine:   Device: laptop System: Dell product: XPS 13 9350
           Mobo: Dell model: 09JHRY v: A00 UEFI: Dell v: 1.4.18 date: 06/06/2017
Battery    BAT0: charge: 12.2 Wh 17.0% condition: 71.6/75.7 Wh (95%) model: SMP DELL JHXPY53 status: Discharging
CPU:       Dual core Intel Core i7-6560U (Skylake rev.3) (-HT-MCP-) cache: 4096 KB
           flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx) bmips: 8834
           clock speeds: max: 3200 MHz 1: 883 MHz 2: 738 MHz 3: 813 MHz 4: 795 MHz
Graphics:  Card: Intel Iris Graphics 540 bus-ID: 00:02.0
           Display Server: x11 (X.Org 1.19.3) driver: intel Resolution: 1280x720@60.00hz
           OpenGL: renderer: Mesa DRI Intel Iris Graphics 540 (Skylake GT3e)
           version: 4.5 Mesa 17.1.8 Direct Render: Yes
Audio:     Card Intel Sunrise Point-LP HD Audio driver: snd_hda_intel bus-ID: 00:1f.3
           Sound: Advanced Linux Sound Architecture v: k4.9.47-1-MANJARO
Network:   Card: Broadcom Limited BCM4350 802.11ac Wireless Network Adapter driver: brcmfmac bus-ID: 3a:00.0
           IF: wlp58s0 state: up mac: <filter>
Drives:    HDD Total Size: NA (-)
           ID-1: /dev/nvme0n1 model: N/A size: 512.1GB
Partition: ID-1: / size: 452G used: 25G (6%) fs: ext4 dev: /dev/nvme0n1p2
           ID-2: swap-1 size: 18.34GB used: 0.00GB (0%) fs: swap dev: /dev/nvme0n1p3
Sensors:   System Temperatures: cpu: 57.0C mobo: 27.8C
           Fan Speeds (in rpm): cpu: N/A
Info:      Processes: 206 Uptime: 2:03 Memory: 3107.5/15898.0MB Init: systemd Gcc sys: 7.1.1
           Client: Shell (bash 4.4.121) inxi: 2.3.27
```
Running Python 3.6.2

Output: 
```
(py-experiments) [beast@beast muse-lsl]$ python3 muse-lsl.py 
[sudo] password for beast: 
Unexpected error when scanning: Set scan parameters failed: Operation not permitted

Traceback (most recent call last):
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pygatt/backends/gatttool/gatttool.py", line 311, in scan
    scan.expect('foooooo', timeout=timeout)
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f6c8f5efa90>
command: /usr/bin/hcitool
args: ['/usr/bin/hcitool', '-i', 'hci0', 'lescan']
buffer (last 100 chars): b''
before (last 100 chars): b'Set scan parameters failed: Operation not permitted\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 1
flag_eof: True
pid: 17629
child_fd: 29
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("b'foooooo'")

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "muse-lsl.py", line 44, in <module>
    muse.connect()
  File "/home/beast/Projects/muse/muse-lsl/muse/muse.py", line 47, in connect
    address = self.find_muse_address(self.name)
  File "/home/beast/Projects/muse/muse-lsl/muse/muse.py", line 68, in find_muse_address
    list_devices = self.adapter.scan(timeout=10.5)
  File "/home/beast/Projects/muse/py-experiments/lib/python3.6/site-packages/pygatt/backends/gatttool/gatttool.py", line 322, in scan
    raise BLEError(message)
pygatt.exceptions.BLEError: Unexpected error when scanning: Set scan parameters failed: Operation not permitted
```

alexandrebarachant, 2017-09-06:
Thanks, i think i have the same error. Looks like the lescan need root privilege.
For me its a pygatt issue. can you open it there ?

luzpaz, 2017-09-06:
@alexandrebarachant i'm not sure what to write in the issue ? Because I don't have a way to reproduce it in a simplified manner. Do you mind opening it instead ?

alexandrebarachant, 2017-09-06:
@luzpaz okay, but i will not be able to do it this week.
if you want to avoid this error, provide the mac address of your device with the `--address` argument.

This error is specific to linux and should not concern people using BLED dongles

luzpaz, 2017-09-06:
@alexandrebarachant np. Sorry to impose.. but do you happen to know how I can determine my device mac address?

alexandrebarachant, 2017-09-06:
@luzpaz no problem.

Just use the command that crashed, but with sudo :) 

`sudo hcitool -i hci0 lescan`

alexandrebarachant, 2017-09-06:
actually, i found a fix ([thanks to the doc of pygatt](https://github.com/peplin/pygatt/blob/3b1865fdd04631905ac3055815d8dcea5183ffa9/pygatt/backends/gatttool/gatttool.py#L297))

> sudo setcap 'cap_net_raw,cap_net_admin+eip' `which hcitool` 

i tried on my laptop, it works

closed 2019-10-22 by jdpigeon