alexandrebarachant/muse-lsl #19: lsl-viewer on mac with anaconda

aj-ptw, 2017-10-22:
Trying to get lsl-viewer-v2.py running.

macOS 10.13
python 3.5

```
/Users/aj/anaconda/envs/future-py35/bin/python /Users/aj/development/muse-lsl/lsl-viewer-V2.py
Traceback (most recent call last):
  File "/Users/aj/development/muse-lsl/lsl-viewer-V2.py", line 15, in <module>
    from seaborn import color_palette
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/seaborn/__init__.py", line 6, in <module>
    from .rcmod import *
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/seaborn/rcmod.py", line 8, in <module>
    from . import palettes, _orig_rc_params
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/seaborn/palettes.py", line 12, in <module>
    from .utils import desaturate, set_hls_values, get_color_cycle
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/seaborn/utils.py", line 11, in <module>
    import matplotlib.pyplot as plt
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/matplotlib/pyplot.py", line 113, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/matplotlib/backends/__init__.py", line 60, in pylab_setup
    [backend_name], 0)
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/matplotlib/backends/backend_macosx.py", line 19, in <module>
    from matplotlib.backends import _macosx
RuntimeError: Python is not installed as a framework. The Mac OS X backend will not be able to function correctly if Python is not installed as a framework. See the Python documentation for more information on installing Python as a framework on Mac OS X. Please either reinstall Python as a framework, or try one of the other backends. If you are using (Ana)Conda please install python.app and replace the use of 'python' with 'pythonw'. See 'Working with Matplotlib on OSX' in the Matplotlib FAQ for more information.

Process finished with exit code 1
```

aj-ptw, 2017-10-22:
Little update.

I did:
```bash
$ conda install matplotlib
```

Which changed the error message of 
```bash
$ python lsl-viewer-V2.py
/Users/aj/anaconda/envs/future-py35/bin/python /Users/aj/development/muse-lsl/lsl-viewer-V2.py
/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/matplotlib/colors.py:680: MatplotlibDeprecationWarning: The is_string_like function was deprecated in version 2.1.
  not cbook.is_string_like(colors[0]):
/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/matplotlib/cm.py:131: MatplotlibDeprecationWarning: The is_string_like function was deprecated in version 2.1.
  if not cbook.is_string_like(name):
looking for an EEG stream...
Start aquiring data
Traceback (most recent call last):
  File "/Users/aj/development/muse-lsl/lsl-viewer-V2.py", line 60, in <module>
    color = color_palette("RdBu_r", nrows)
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/seaborn/palettes.py", line 177, in color_palette
    palette = mpl_palette(palette, n_colors)
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/seaborn/palettes.py", line 413, in mpl_palette
    palette = list(map(tuple, cmap(bins)[:, :3]))
  File "/Users/aj/anaconda/envs/future-py35/lib/python3.5/site-packages/matplotlib/colors.py", line 487, in __call__
    cbook._putmask(xa, xa == 1.0, almost_one)
AttributeError: module 'matplotlib.cbook' has no attribute '_putmask'

Process finished with exit code 1
```

them4ra, 2017-10-23:
What version of NumPy do you have? You can try an update and see if that works since 'putmask' is defined in 'numpy.core.multiarray'.

aj-ptw, 2017-10-23:
@them4ra i'm using `numpy` 1.13.3

them4ra, 2017-10-24:
Ok, in that case it probably isn't a thing with Numpy. Just to get on the same page, I'm guessing you have all the [requirements](https://github.com/alexandrebarachant/muse-lsl/blob/master/requirements.txt) installed and running? If that is the case, I'm guessing it might have to do with your version of matplotlib and how it's interacting with your version of Anaconda.

closed 2019-10-22 by jdpigeon