alexandrebarachant/muse-lsl #22: No BGAPI compatible device detected on windows10

ranibourbara, 2017-11-27:
Hi

I have purchased a BLED112 dongle and downloaded the SDK from bluegiga. yet I keep getting
 No BGAPI compatible device detected, when I run muse-lsl.py
(note that I disabled the internal bluetooth of my laptop)
any directions on what is the problem?

alexandrebarachant, 2017-11-27:
okay, so if you have installed the driver and everything else (pygatt) correcly, then it might be a problem with the auto detection of the dongle.
try to identify the serial port associate with the device and specify it to the code. For example :

```python muse-lsl.py --adress myaddress --interface COM9```

otherwise, try to check if the issues on the pygatt repository : https://github.com/peplin/pygatt/issues

ranibourbara, 2017-11-27:
Thank you Alex for your prompt response.
It is now working like a charm.
The problem  ,for people experiencing this, was related to the pygatt. 
Simply follow https://github.com/peplin/pygatt/issues/118 , specifically eliquious  comment.
It appears that on order to establish a connection the code needs to enable multiple trials to connect.

Thank you

alexandrebarachant, 2017-11-27:
Great.
I'm keeping this issue open so that people can see the solution.
I hope they will fix it soon

closed 2019-10-22 by jdpigeon