alexandrebarachant/muse-lsl #23: Get Data Accelerometer & Gyroscope

TheChapu, 2017-12-12:
Hello,
I'm trying to get the data from the Accelerometer and Gyroscope, to add it to the other data

With an Android app, I inspected the services and found the ones that were used for the EEG Data

#EGG
273e0003-4c4d-454d-96be-f03bac821358
273e0004-4c4d-454d-96be-f03bac821358
273e0005-4c4d-454d-96be-f03bac821358
273e0006-4c4d-454d-96be-f03bac821358
273e0007-4c4d-454d-96be-f03bac821358

and these are "Free"

#Books
273e000A-4c4d-454d-96be-f03bac821358
273e000B-4c4d-454d-96be-f03bac821358
273e0001-4c4d-454d-96be-f03bac821358
273e0002-4c4d-454d-96be-f03bac821358
273e0008-4c4d-454d-96be-f03bac821358
273e0009-4c4d-454d-96be-f03bac821358

Someone here has managed to get the other data

alexandrebarachant, 2017-12-12:
Hey, 
i think @urish decoded the Accelerometer data for his muse-js repository : https://github.com/urish/muse-js
I will be more than happy if you submit a PR for this :)

TheChapu, 2017-12-12:
Great,

I'll see how to pass the code to Python.

I'll do tests, if everything goes well, I'll upload it here.

I hope everything goes well

TheChapu, 2017-12-12:
# I was working a bit and this got,

muse_service = 0xfe8d
 control_characteristic = 273e0001-4c4d-454d-96be-f03bac821358
 telemetry_characteristic = 273e000b-4c4d-454d-96be-f03bac821358
 gyroscope_characteristic = 273e0009-4c4d-454d-96be-f03bac821358
 accelerometer_characteristic = 273e000a-4c4d-454d-96be-f03bac821358

 egg_characteristic = 273e0003-4c4d-454d-96be-f03bac821358,
					   273e0004-4c4d-454d-96be-f03bac821358,
						273e0005-4c4d-454d-96be-f03bac821358,
						273e0006-4c4d-454d-96be-f03bac821358,
						273e0007-4c4d-454d-96be-f03bac821358,
 egg_frequency = 256
 eeg_characteristic = TP9,AF7,AF8,TP10,AUX
 backend = 'gatt'
 pygatt.GATTToolBackend(self.interface)
 interface = 'hci0'

`
import bitstring
import pygatt
import numpy as np
from time import time, sleep
from sys import platform

class Muse():
	
    """Muse 2016 headband"""

    def __init__(self, address=None, callback=None, eeg=True, accelero=True,
                 giro=True,telemetry=True,info=True, backend='auto', interface=None, time_func=time,
                 name=None):
        """Initialize"""
        self.address = address
        self.name = name
        self.callback = callback
        self.eeg = eeg
        self.accelero = accelero
        self.giro = giro
        self.telemetry = telemetry
        self.info = info
        self.interface = interface
        self.time_func = time_func

        if backend in ['auto', 'gatt', 'bgapi']:
            if backend == 'auto':
                if platform == "linux" or platform == "linux2":
                    self.backend = 'gatt'
                else:
                    self.backend = 'bgapi'
            else:
                self.backend = backend
        else:
            raise(ValueError('Backend must be auto, gatt or bgapi'))

    def connect(self, interface=None, backend='auto'):
        """Connect to the device"""

        if self.backend == 'gatt':
            self.interface = self.interface or 'hci0'
            self.adapter = pygatt.GATTToolBackend(self.interface)
        else:
            self.adapter = pygatt.BGAPIBackend(serial_port=self.interface)

        self.adapter.start()

        if self.address is None:
            address = self.find_muse_address(self.name)
            if address is None:
[... 197 lines cut ...]
        # last data received
        #Si el ultimo dato es = 35
        if handle == 35:
			#Si el tiempo es distinto
            if tm != self.last_tm + 1:
                print("missing sample %d : %d" % (tm, self.last_tm))
            self.last_tm = tm

            # calcultate index of time samples
            # Calcula el indice de muestras en el tiempo
            idxs = np.arange(0, 12) + self.sample_index
            self.sample_index += 12

            # affect as timestamps
            # Afeecta a timestamp
            timestamps = self.reg_params[1] * idxs + self.reg_params[0]

            # push data
            # Se envia la data a la funcion #callback
            self.callback(self.data, timestamps)
            self._init_sample()
`

Build 2 new functions to obtain Acc data, Telemetry data
I'm still not sure if the data I get is correct

luzpaz, 2017-12-24:
Is this ready to be submit as a PR ?

alexandrebarachant, 2017-12-25:
hey, @urish made some progress on decoding gyroscope and accelero data : https://github.com/urish/muse-js/commit/66a8eba78570300e580c9ae47d379e7f9d6ddc8b

@TheChapu can you please make a [pull request](https://github.com/PointCloudLibrary/pcl/wiki/A-step-by-step-guide-on-preparing-and-submitting-a-pull-request) with your changes so i can try and review your code.

TheChapu, 2017-12-26:
hey @alexandrebarachant 
I have created a repository so as not to spoil yours :F

Please review. https://github.com/TheChapu/Muse-Python

I have obtained the data of the Accelerometer and Gyroscope, taking as an example the code of @urish 

Please check it and tell me to update

alexandrebarachant, 2017-12-26:
great, the intent is good, but this way I can not check changes and merge your modifications easily (it will require a lot of extra manual work).

The usual way to do this on github is to : 
1. fork the repository (i.e. create a copy in your own account) by using the fork button
2. create a new branch with your modifications
3. submit a pull request.

TheChapu, 2017-12-26:
[image]

alexandrebarachant, 2017-12-26:
almost there. you have to submit the pull request to my repository (not yours)

TheChapu, 2017-12-26:
Sorry.

Can you look again?

Thank you

alexandrebarachant, 2017-12-27:
hey, 

go to this page : https://github.com/alexandrebarachant/muse-lsl/pulls

click on 'new pull request', then select your brqnch qnd submit it.

TheChapu, 2017-12-28:
[image]

TheChapu, 2017-12-28:
I see the image.

I'm sorry, but it's the first time I use Github to help

alexandrebarachant, 2017-12-28:
you probably have to click on "compare across forks" then select your branch containing your modification and my master branch.

don't worry, we all had to make our first PR someday :)

TheChapu, 2017-12-28:
At last :+1: 
I won a coffee  

[image]
[image]

jdpigeon, 2019-05-17:
Address by https://github.com/alexandrebarachant/muse-lsl/pull/93

closed 2019-05-17 by jdpigeon