alexandrebarachant/muse-lsl #33: How to solve "missing sample " problems ?

hoikc, 2018-01-30:
Thank you for your awesome library for MUSE!

I got raw data from MUSE successfully in my raspberry pi.

However i suddenly got missing sample, since then it countinues. 

According to your code, it seems that it is problem of pygatt or MUSE itself in my thought.

Is there anyone who get this problems?

alexandrebarachant, 2018-01-30:
Did you find a solution ? 
You will get missing sample when the BLE connection drop packets, and there is not much to do about it. We could replace them with zeros, or nans ...

hoikc, 2018-01-31:
Hello, Alexandre !

First of all, thank you for your kind reply and awesome library for MUSE.

I solved it, but I have a question about it.

I used muse-lsl and socket communication to send data from raspberrypi to
pc.

I got "missing sample" when I put socket communication part in muse-lsl.py.

In my thought, missing sample occrus when Streaminlet and Streamoutlet are
in one terminal.

So I run muse-lsl.py and socket communication in another terminals, then
there's no missing sample.

I think that missing sample occurs when bluetooth communication fails
between muse.

Is it a problem that using Streaminlet and Streamoutlet in one script?

Thank you.

Here's my script.

from muse import Muse
from time import sleep
from pylsl import StreamInfo, StreamOutlet, local_clock, StreamInlet,
resolve_byprop
from optparse import OptionParser
import socket
import pickle
import numpy as np
import time

parser = OptionParser()
parser.add_option("-a", "--address",
                  dest="address", type='string',
default='00:55:DA:B3:4C:2D',
                  help="device mac address.")
parser.add_option("-n", "--name",
                  dest="name", type='string', default=None,
                  help="name of the device.")
parser.add_option("-b", "--backend",
                  dest="backend", type='string', default="auto",
                  help="pygatt backend to use. can be auto, gatt or bgapi")
parser.add_option("-i", "--interface",
                  dest="interface", type='string', default=None,
                  help="The interface to use, 'hci0' for gatt or a com port
for bgapi")

(options, args) = parser.parse_args()

info = info = StreamInfo('Muse', 'EEG', 5, 256, 'float32',
                         'Muse%s' % options.address)

info.desc().append_child_value("manufacturer", "Muse")
channels = info.desc().append_child("channels")

for c in ['TP9', 'AF7', 'AF8', 'TP10', 'Right AUX']:
    channels.append_child("channel") \
        .append_child_value("label", c) \
        .append_child_value("unit", "microvolts") \
        .append_child_value("type", "EEG")

outlet = StreamOutlet(info, 12, 2400)

def process(data, timestamps):
    for ii in range(12):
        outlet.push_sample(data[:, ii], timestamps[ii])

muse = Muse(address=options.address, callback=process,
            backend=options.backend, time_func=local_clock,
            interface=options.interface, name=options.name)

def data_put(threadname, q):
    while True:
        sample, timestamp = inlet.pull_chunk(timeout=1.0, max_samples=12)

muse.connect()
print('Connected')
muse.start()
print('Streaming')

buf = 12
        %%%% I take this part for lsl-viewer
streams = resolve_byprop('type', 'EEG',timeout=2)

if len(streams)==0:
    raise(RuntimeError("Can't find EEG stream!"))
inlet = StreamInlet(streams[0],max_chunklen=buf)

s = socket.socket()
host = '192.168.137.213'
port = 12345
s.bind((host,port))
s.listen(5)

c,addr = s.accept()
print ('Got connectionfrom', addr)

while 1:
    try:
        sample, timestamp = inlet.pull_chunk(timeout=1.0, max_samples=12)
        c.send(pickle.dumps(sample))
        c.recv(1024)

    except:
        c.close()
        s.close()
        muse.stop()
        muse.disconnect()
        print('Disonnected')

maxihidara, 2018-11-14:
Is this problem solved with hoikc's solution ?

jdpigeon, 2018-11-14:
I believe so. However, this is an outdated issue that I wouldn't recommend basing any new code off of.

If you're running into issues using muse-lsl through the higher-level API (e.g. stream, view), then I'd recommend creating a new issue

closed 2018-11-14 by jdpigeon