alexandrebarachant/muse-lsl #44: lsl-viewer.py gets stuck in self.fig.canvas.draw()

emanuelgo, 2018-05-25:
Hi, I am running lsl-viewer.py on Mac 10.11, python 2.7, conda 4.4.10, muse 2016
my lsl stream works and lsl-viewer-V2.py as well.

everytime when i open lsl-viewer.py it gets stuck at  self.fig.canvas.draw() 
sometimes i can see a second of eeg data plotted, usually just an empty canvas  
because there were no errors visible (it just got stuck) i added some print("..") markers to follow the process and my print("..") markers  suggest that it gets stuck after self.fig.canvas.draw()
usually in the first run, sometimes in the second..

Any suggestions for getting this solved?

emanuelgo, 2018-05-25:
same issue in python 3.6

jdpigeon, 2018-06-01:
Try deleting line 202: `lslv.stop()` and see if that allows the viewer to run. I was playing around with the running the viewer from a Jupyter Notebook and needed to make that change to get around a very similar error

emanuelgo, 2018-06-01:
unfortunately deleting line 202 didn´t work for me. but i found out if I delete self.fig.canvas.draw() in line 156, i can use the reset button placed in the submenu do manually draw the current data stream 1 frame per click as fast as i click ... what do you think could that mean?

jdpigeon, 2018-06-04:
Take a look at this issue, which I think might be what you're experiencing: https://stackoverflow.com/questions/3441874/matplotlib-animation-either-freezes-after-a-few-frames-or-just-doesnt-work

The loop logic in lsl viewer is very simple and doesn't take advantage of the GUI-specific animation calls those commenters are recommending. Thus, it may be susceptible to freezy and laggy behaviour that varies between platforms. We've also noticed a lot of viewer lag on a few Windows machines that could be related to this as well. 

We'll look into rewriting the looping code in the future. For the meantime, could you try using a different GUI backend? Try both `matplotlib.use('TkAgg')` and `matplotlib.use('GTKAgg')`

Also, does lsl_viewer_v2 work for you? You may have to install vispy, but it works better on some machines

jdpigeon, 2018-06-05:
Tested this out myself on a Windows machine:

The default viewer_v1 worked terribly. It was slow and crashed every time I tried to move the window. However, viewer_v2 worked very well.

I was able to improve the performance of viewer_v1 immensely by explicitly setting the backend to 'TkAgg'. This tip is corroborated here, too: https://www.swharden.com/wp/2013-04-15-fixing-slow-matplotlib-in-pythonxy/

I went back to my Linux machine and tested using 'TkAgg' there as well. There's no measurable effect on performance there, so I'd like to make this explicit call default in https://github.com/kowalej/muse-lsl/commit/33f7d3943a7a5d0b5f7706b734b23501df485eb6. We can revisit it later on if this causes issues in other platforms

closed 2019-10-22 by jdpigeon