alexandrebarachant/muse-lsl #49: list_muses() always returns None on Windows

crocodoyle, 2018-06-15:
Not clear how this is supposed to work

jdpigeon, 2018-06-15:
Unless the bgapi backend is specified `list_muses` will always just start the BlueMuse app on Windows 10. I've update the README to hopefully make this more clear.

If this becomes an issue for more people we should consider removing these BlueMuse startup triggers in the CLI and making it default to bgapi. That may reduce the cross-platform complexity, especially when importing Muse LSL functions in separate projects

tsloan1377, 2019-06-07:
Having the same issue on Windows 10 using a Jupyter Notebook.
Following usage as a package:

```
from muselsl import stream, list_muses, view
muses = list_muses()
```
Starting BlueMuse, see BlueMuse window for interactive list of devices.
`display(muses)`
None

How should I specify bgapi as a backend?

Just to clarify, BlueMuse loads and runs succesfully, remaining streaming to my Muse, but I'm unable to access with Python.

jdpigeon, 2019-06-11:
There currently isn't any communication between muselsl and BlueMuse available beyond muselsl starting BlueMuse and BlueMuse opening up an LSL stream that muselsl can record and view from. So, list_muses returning `None` on Windows without specifying the bgapi backend `muselsl list -b bgapi` is intentional.

However, not being able to connect to an existing BlueMuse stream with `muselsl record` or `muselsl view` is an issue. Let me know if that issue arises.

closed 2019-10-22 by jdpigeon