alexandrebarachant/muse-lsl #51: listing muses on Mac

amandakeasson, 2018-06-16:
I get the following error when I run either muselsl list from the command line or list_muses() in python on my Mac (Sierra):

Searching for Muses, this may take up to 10 seconds...
Unexpectedly not connected to USB device
Traceback (most recent call last):
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/bin/muselsl", line 11, in <module>
    sys.exit(main())
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/muselsl/cli.py", line 50, in __init__
    getattr(self, args.command)()
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/muselsl/cli.py", line 63, in list
    list_muses(args.backend, args.interface)
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/muselsl/stream.py", line 27, in list_muses
    devices = adapter.scan(timeout=10000)
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 301, in scan
    scan_interval, scan_window, parameters
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 248, in send_command
    raise NotConnectedError()
pygatt.exceptions.NotConnectedError

jdpigeon, 2018-06-16:
Can you let me know if the list_muses function works when the bgapi backend and COM port that the BLED112 dongle is connected to is specified?

`list_muses(backens='bgapi', interface='COM9')`

amandakeasson, 2018-06-17:
I added "self._open_serial_port()" to the BGAPIBackend scan function so that the port doesn't have to be entered manually. That part worked,  but now I get this error: 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/muselsl/stream.py", line 27, in list_muses
    devices = adapter.scan(timeout=10)
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 308, in scan
    self.expect(ResponsePacketType.gap_set_scan_parameters)
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 525, in expect
    return self.expect_any([expected], *args, **kargs)
  File "/Users/amandae/anaconda2/envs/neurobrite_test2/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 564, in expect_any
    raise exc
pygatt.backends.bgapi.exceptions.ExpectedResponseTimeout: Timed out after 10.000000s waiting for [<ResponsePacketType.gap_set_scan_parameters: 64>]

jdpigeon, 2018-06-17:
So, specifying the backend and interface didn't change the errors you're getting?

The reason I suggested that is because we didn't actually change any of the pygatt code in the lastest update, just stuff around pygatt such as our [resolve_backend](https://github.com/alexandrebarachant/muse-lsl/blob/master/muselsl/helper.py) function, which we unfortunately weren't able to test on a Mac.

Regarding the `open_serial_port` function, it looks like it's called automatically by the [BGAPIBackend init function](https://github.com/peplin/pygatt/blob/79f69acce401cf9a204bc320bfc9a72f24230252/pygatt/backends/bgapi/bgapi.py#L76) and shouldn't need to be called again. I don't know too much about the inner workings of pygatt and am not able to advise about the duration.

Based on the errors you're getting it looks like there's some problems with pygatt or the BLED dongle's connectivity on your system. Just as a sanity check, could you see whether this simple script from a year ago works on your system currently? https://github.com/alexandrebarachant/muse-lsl/blob/7c342b72fa553dbe3c8e4038eb7cc77fd825b98b/scan_devices.py

amandakeasson, 2018-06-18:
I was hoping to try to figure out how to get it working without having to specify the backend and interface. When I do specify them, I still get the same timeout error.

I managed to get it working though- we just need to add "adapter.start()" before "adapter.scan(timeout=MUSE_SCAN_TIMEOUT)" in stream.py.

jdpigeon, 2018-06-18:
Great news! 🎉 

It looks like calling `adapter.scan` is a special requirement when using the BGAPI backend that we didn't account for: https://github.com/peplin/pygatt/blob/b2777f3b781a71e3a7611b88d9a87f8b4c177466/pygatt/backends/bgapi/bgapi.py#L81

We should make a PR soon that introduces start (and stop) calls smartly into the list_muses function. It's already included in the Muse class which handles connection.

amandakeasson, 2018-06-19:
Sounds good. Thanks for your help!

closed 2018-06-19 by jdpigeon