alexandrebarachant/muse-lsl #56: GLIBC error

cedricsimar, 2018-07-05:
Hello,

First, thanks for all your work that is a huge help for EEG enthusiasts (and occasionally Kaggle competitors :-))

I run Ubuntu 14.04 with Python 3.6 and when I import muse-lsl I get the following error message: 
"OSError: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.25' not found (required by /home/cml/anaconda3/envs/ml/lib/python3.6/site-packages/pylsl/liblsl64.so)"

Do you know if there is an straightforward way to fix this on Linux or should I move to Windows?

Thanks in advance for your help,

Cédric.

jdpigeon, 2018-07-05:
I think this is because your Ubuntu version is older than what pylsl has been compiled to run on.

I'd recommend trying to update glibc: https://askubuntu.com/questions/736208/upgrading-glibc

If that doesn't work, Ubuntu 18 is really nice ;)

cedricsimar, 2018-07-06:
Thank you for your reply, I'll give another shot at updating glibc then 
Otherwise I'll follow your advise and give up to Ubuntu 18 (or worse, Windows :-)) depending on what they prefer in the lab.