alexandrebarachant/muse-lsl #59: `muselsl` connects, but does not stream data on ubuntu

ysimonson, 2018-08-04:
Hi, first of all thanks for the project! There's not much out there on getting streaming data from 2016 headsets, so it's much appreciated.

I'm using ubuntu + BLED112 and trying to stream EEG data. I can get to the point of pairing/connecting, but `muselsl` automatically disconnects because it doesn't get any data:

```
$ muselsl stream --address 00:55:DA:B0:8E:E7 --backend bgapi
Connecting to Muse : 00:55:DA:B0:8E:E7...
Connected.
Streaming...
Timestamp 1533395995.9020605
^CTraceback (most recent call last):
  File "/home/ysimonson/skull/venv/bin/muselsl", line 11, in <module>
    sys.exit(main())
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/muselsl/__main__.py", line 51, in main
    cli = CLI(args.command)
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/muselsl/cli.py", line 9, in __init__
    getattr(self, command)()
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/muselsl/cli.py", line 42, in stream
    args.interface, args.name)
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/muselsl/stream.py", line 116, in stream
    muse.stop()
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/muselsl/muse.py", line 204, in stop
    self._write_cmd([0x02, 0x68, 0x0a])
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/muselsl/muse.py", line 123, in _write_cmd
    self.device.char_write_handle(0x000e, cmd, False)
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/device.py", line 21, in wrapper
    return func(self, *args, **kwargs)
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/device.py", line 116, in char_write_handle
    EventPacketType.attclient_procedure_completed)
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 521, in expect
    return self.expect_any([expected], *args, **kargs)
  File "/home/ysimonson/skull/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 557, in expect_any
    raise exc
pygatt.backends.bgapi.exceptions.ExpectedResponseTimeout: Timed out after 1.000000s waiting for [<EventPacketType.attclient_procedure_completed: 17>]
```

As far as I've seen, this hasn't come up before with linux + BLED112 since it's the blessed solution. Is there something obvious I'm missing?

ysimonson, 2018-08-06:
That yields this error: `pygatt.exceptions.BLEError: No BLE adapter found`.

When I first got that error, I thought I might be missing drivers. But AFAICS none need to be manually installed. I switched to bgapi as a workaround - it successfully finds the adapter, and even pairs, but then fails to stream data.

jdpigeon, 2018-08-06:
Hmm, that's odd.

Unfortunately, I'm all out of Muse LSL-specific advice to give you. I think your issues definitely lie with pygatt itself and I recommend checking out their Issues section on Github. 

Some quick thoughts:
1. The errors you're getting when using BGAPI might be related to this previously reported but unresolved issue: https://github.com/peplin/pygatt/issues/84
2. If you want to go the route of using the default Linux backend, I would make sure you have the most recent version of BlueZ, gatttool, and pexpect installed