alexandrebarachant/muse-lsl #61: Multiple eeg streams simultaneously

bhager2127, 2018-08-22:
Trying to view two simultaneously connected muse bands via muselsl. Freezes the program whenever this is attempted. Both bands are detected. Advice? 

p.s. Using windows 10 and BlueMuse

jdpigeon, 2018-08-22:
I've encountered this problem too. I'm worried that it might be related to limitations of the bluetooth adapter BlueMuse uses. Thoughts, @kowalej?

For now, the recommended solution for multiple EEG streams is using Muse Direct, but that only supports OSC protocol streaming.

kowalej, 2018-08-23:
Tested this tonight on 2 different machines with BlueMuse version 1.0.7.0 and 1.0.8.0 - both worked. First machine was a customer gaming PC with onboard Bluetooth chip (Qualcomm Atheros), other is an older, cheap Dell laptop, again with built in Bluetooth (also manufactured by Qualcomm Atheros).

Which version of BlueMuse are you using? It may be something that was fixed in one of the later versions.

rschmaelzle, 2019-01-08:
@kowalej Hi Kowaley - that sounds great. I've been tinkering a bit with this as well, but somehow my skills aren't enough to figure out code-wise how to connect 2 muses with 1 machine. I am currently using a mac and have the mac-addresses of both, but I don't really understand how to get the streams started. 
`stream_process = Process(target=stream, args=( muses[my_muse_index-1]['address'],) )`
If I submit both mac-addresses as arguments to the Process, it complains (because the muselsl-function stream seems designed differently, but if I start two separate stream_process (1 &2), either one or both will crash. I will also try using bluemuse, but perhaps there's an obvious way to use the multiprocess.process ?
Any hint would help. Best, r

kowalej, 2019-01-09:
@rschmaelzle not sure about the issue with your Mac in regards to streaming with muse-lsl directly, but just a heads up - BlueMuse works only for Windows.

rschmaelzle, 2019-01-09:
thanks, will try from windows using bluemuse! if anyone else has an idea how to do via the process and stream method, that would be great.

DerekLomas, 2019-03-07:
Did this work for you? Has anyone gotten multiple devices to work with Muse2? Should there be any difference? I'm trying to decide which device to get for a hyperscanning study. Thanks!

Kethasi, 2019-05-07:
I would also be interested whether muse-lsl will allow me to record from multiple Muse 2 devices to one Mac. Many thanks for any hints on this!

jdpigeon, 2019-05-16:
@DerekLomas @Kethasi , One thing you could consider is making use LSL's [network capabilities](https://github.com/sccn/labstreaminglayer/wiki/NetworkConnectivity) to set up multiple EEG streams shared between computers. You'll need a separate laptop for each Muse, but once they're all connected via bluetooth their LSL streams should be discoverable by any computer on the local network for simultaneous processing

rschmaelzle, 2021-07-15:
as an update. 2 Muses work fine with recent versions of BlueMuse. There can be crashes, but it is definitely possible - so if anyone is wondering/trying, it makes sense to keep at it.

I have one follow-up question: NOw that I have two streams, I would love to "merge" or "combine" them. I wouldn't mind about possible smaller time-jitters (I know LSL could correct them) for now, but would just like to do something like 
s1 = streams[0].get_data(last_100_samples)
s2 = streams[1].get_data(last_100_samples)
merged = np.vcat)[s1,s2])
plt.plot(merged)

The code may not be correct, but hopefully you get the gist. If anyone had ideas on how to accomplish that, i'd be very grateful. 
Best, r