alexandrebarachant/muse-lsl #66: Acc, Gyro and Telemetry in recording output

Dutchflag, 2018-11-11:
Hi,

many thanks for all the code so far! 

I've managed to get muselsl working on a Raspberry Pi 3 model B+. With the CLI record function, I get the raw EEG + timestamps in a .csv (from a Muse 2016) .
In muse.py, i noticed that all the code for subscribing to acc, gyro and telemetry is there, and also the handling of the resulting data.

Is there a reason why this is not added to the recording (yet)? I'd like to use it for automated artefact classification (in order to get rid of epochs movement artefacts, besides eye blinks & jaw clenches).

As a side note: have you played with the Muse-2 already?

jdpigeon, 2018-11-12:
Hey Dutchflag!

You're right, it would be nice to have acc, gyro, and telemetry output, and there's no reason not to have it.

We actually have two outstanding PRs waiting to be merged that introduce those data types. I have just been sidetracked by other side projects and haven't found the time to go over them and continue the work by adding the other data streams into the CLI. I'll try to carve out some time next weekend.

In the meantime, feel free to go ahead and use the code in those PR branches if you want to get those data types. You just won't be able to use the CLI. https://github.com/alexandrebarachant/muse-lsl/pull/34

Dutchflag, 2018-11-12:
tnx, I will, I have to use it in a python script anyway. Just wanted to check if the new functionality was already finished and tested ;-). I suppose the interpretation of the data as been verified by urish in his muse-js library?

I'll keep you posted.

jdpigeon, 2019-05-17:
Address by https://github.com/alexandrebarachant/muse-lsl/pull/93

closed 2019-05-17 by jdpigeon