alexandrebarachant/muse-lsl #67: Getting "missing sample" ?

maxihidara, 2018-11-14:
Hi, i am usually getting "missing sample" when stream and run exercise 2 from bci workshop python file. I have tried the solution from previous issue but get error when i run given script, and i don't really understand about it. I am using raspberry pi 3 and muse 2016. Can anyone help ? Thank  you

maxihidara, 2018-11-15:
Is using another bluetooth adapter (not raspi built in bluetooth adapter) fix the problem ?

Nope, never happens to me that issue

jdpigeon, 2018-11-15:
Yea, if you can get a BlueGiga BLED112 dongle, that might be able to solve your issues

maxihidara, 2018-11-16:
Is there another bluetooth dongle that compatible with muse eeg ? Because i can find online seller who sell it in my country.

maxihidara, 2018-11-16:
I mean other than bluegiga bled112

jdpigeon, 2018-11-16:
There may be, but we've encountered lots of dongles that don't work. Our strategy is just to recommend BlueGiga BLED112. If you talk to a hardware specialist, they could hopefully point you to one that may be compatible

maxihidara, 2018-11-19:
I just check my raspberry's bluetooth specification, it says that my raspberry pi 3 model b, using 4.1 bluetooth version, and if i am not wrong, muse eeg 2016 using 4.0 bluetooth.

maxihidara, 2018-11-20:
I have solved it. I use eksternal bluetooth dongle (not bluegiga). It's right what jdpigeon said that raspberry pi bluetooth have bad performance

closed 2018-11-20 by maxihidara