alexandrebarachant/muse-lsl #68: muselsl via BLE112 not streaming on Windows 10

mgpritchard, 2018-11-16:
Hiya, I'm running on python3.7, 32bit Win10 on a pretty old laptop that doesn't have a built-in bluetooth receiver so am using a BLE112 dongle and the bgapi backend.

muselsl will happily find the device but when I start a stream it's timing out almost immediately, looks like its just not getting any data through:

` $ muselsl stream -b bgapi --address 00:55:DA:B3:9A:2C `
` c:\users\michael\appdata\local\programs\python\python37-32\lib\site-packages\sklearn\externals\joblib\externals\cloudpickle\cloudpickle.py:47: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses `
`  import imp `
`Connecting to Muse : 00:55:DA:B3:9A:2C...`
`Connected.`
`Streaming...`
`Disconnected. `

Seems like a very similar issue to #59 
Any advice?

jdpigeon, 2018-11-16:
This is really interesting, and makes me think that muse-lsl's current dependencies might not be working with the bgapi backend.

Take a look at this thread and the corresponding fork and dependencies file that @amandakeasson recommends: https://github.com/NeuroTechX/eeg-notebooks/issues/3

If that helps get data streaming on your computer, please post back here and we can update the dependencies for everyone!

mgpritchard, 2018-11-19:
Thanks for the advice, that does seem to have done the trick!
Running now in a miniconda environment with python 3.6.7, and the dependencies from https://github.com/amandakeasson/eeg-notebooks/blob/master/requirements_mac.txt

Now getting data through! The stream does report:
`missing sample 0 : 65535`
but the stream no longer kicks out and is definitely receiving data:
[image]

For completeness I should mention that I've also implemented the `time.sleep(.25)` fix referenced in peplin/pygatt/issues/118. I'd already implemented that on Friday when the stream still wasn't working so it seems to be a wholly separate issue to do with pygatt accessing the COM port, but worth mentioning.

closed 2018-11-19 by mgpritchard

mgpritchard, 2018-11-20:
No worries, couldn't have done it without your suggestion
If I had the free time I'd go through and try to establish which individual dependencies were specifically at fault... Something for a rainy day perhaps

rschmaelzle, 2019-06-03:
hello, i am not sure if my error is related. I am - after running amandakeasson's mac-specific adjustments - able to run the notebooks to connect the muse. then, in a separate terminal, i am also able to run muselsl view, check the signals, and finally run sample experiments. 
however, I now tried to make everything more automatic, and particularly I tried to run the code to connect the muse in a simple python file (code  see below - or I can  run 'muselsl stream' alone from the commandline). However, whenever I leave the jupyter notebooks, the muse does initially connect ("streaming..."), but then after about 0.5 sec it says "Disconnected". Any ideas why it runs smoothly in the notebooks but not in the terminal or via a python script (e.g. python3 connect_muse.py)?

Best, r

KEy code that is in the jupyter notebook or the python file:
`
from muselsl import stream, list_muses, view, record
from multiprocessing import Process
muses = list_muses()

stream_process = Process(target=stream, args=(muses[0]['address'],))
stream_process.start()`