alexandrebarachant/muse-lsl #71: Muse 2 support

berkott, 2018-12-25:
Is it possible to use the Muse 2 with this library and if not is there anything else I can use to get all of the data (breath, heart, eeg) emitted by the muse 2?

compmonks, 2019-01-15:
I just purchased a Muse 2 in 2018 and after running into a few bugs with the Bled112, the following fix for pygatt solved the problem : [#119](https://github.com/peplin/pygatt/issues/119)

Currently tested on masOC mojave 10.14.2 and python 3.5

Dutchflag, 2019-01-15:
That is great news compmonks... were you able to get raw EEG or also other data like accelero?

compmonks, 2019-01-15:
for now just quick tests on raw eeg but I will be happy to post more issues or fixes along the way :)

Dutchflag, 2019-01-27:
This also relates to issue #66 . The muse.py code has all the functions for the other data (telemetry, accelerometers, gyro etc) of the Muse2016. However, this is not utilised yet in recording or viewing with muselsl, i am trying to work this out.

I can imagine that Muse2 reuses the same GATT attributes for telemetry, accelero and gyro, which means the corresponding data will be relatively easy to read. This of course needs to be tested with a muse2.

The new sensors of the muse2 (ppg) will have new GATT attributes though, and i haven't seen any documentation on that yet.

HashemiScience, 2019-01-29:
I have some extra info re: the sensors in Muse 2 for those that are better able to navigate through the muselsl code than I.
1. there's no longer a gyroscope in Muse 2
2. the PPG sensor has 3 channels:
- channel 0: ambient light, channel 1: infrared, and channel 2: red
I tried to play with the code in #34, but am not proficient enough in python to adapt it.

Dutchflag, 2019-02-08:
As a first step I wanted to rewrite the record_direct function a bit, to save accelero data directly to a .csv file. This works now for me with a Muse2016 on a Raspberry Pi. 
If anyone is interested, i can share the code.

Next week i will test this on a Muse 2.

Dutchflag, 2019-02-14:
did some early tests with the muse 2 via MuseMonitor -on my iPhone- and muselsl on the RPi. 

I saw both accelerometer and gyro signals via MuseMonitor although they seem to have switched GATT attributes; something strange is going on there...

@HashemiScience ; the specs on https://choosemuse.com/muse-2/ also say that breath (frequency) is deducted from PPG + gyroscope, so i think the gyroscope is still in?

HashemiScience, 2019-02-14:
@Dutchflag you're right. Not sure why one of their devs told me otherwise... I must have misheard!

Dutchflag, 2019-02-22:
after further analysis i think that muse2016 and muse2 are identical for EEG , GYRO, and ACC signals. Only with telemetry i saw some differences; i got a temperature reading of 0 with the muse 2 . 

@HashemiScience : do you have any idea on how the 3 PPG channels are used? Maybe FFT analysis and combine the results of the channels?

pskorupinski, 2019-02-27:
The README file states:

> **Only compatible with the 2016 model of the Muse headband**

Should I assume this to be incorrect?

jdpigeon, 2019-02-28:
Yes, let's change this

closed 2019-02-28 by jdpigeon