alexandrebarachant/muse-lsl #72: I can list Muses but not stream data - suggestions?

MuseUK, 2019-01-15:
Hi - I'm able to see the MAC address of my Muse headset using "muselsl list", but get the following error when I try and stream data. Any ideas?

Unexpected error during construction of stream outlet: lslboost::condition_variable::condition_variable() constructor failed in pthread::cond_init: Invalid argument
Traceback (most recent call last):
  File "/home/brix2/.local/bin/muselsl", line 11, in <module>
    sys.exit(main())
  File "/home/brix2/.local/lib/python3.5/site-packages/muselsl/__main__.py", line 51, in main
    cli = CLI(args.command)
  File "/home/brix2/.local/lib/python3.5/site-packages/muselsl/cli.py", line 9, in __init__
    getattr(self, command)()
  File "/home/brix2/.local/lib/python3.5/site-packages/muselsl/cli.py", line 42, in stream
    args.interface, args.name)
  File "/home/brix2/.local/lib/python3.5/site-packages/muselsl/stream.py", line 80, in stream
    outlet = StreamOutlet(info, LSL_CHUNK)
  File "/home/brix2/.local/lib/python3.5/site-packages/pylsl/pylsl.py", line 409, in __init__
    raise RuntimeError("could not create stream outlet.")
RuntimeError: could not create stream outlet.

jdpigeon, 2019-01-15:
Hmm, sorry you're running into this.

This looks to me like a C++ error coming from the lib-lsl library. I couldn't find any info on the internet about people having similar issues, so this may be a new bug. I'd recommend making an issue on the official labstreaminglayer respository here: https://github.com/sccn/labstreaminglayer/issues?q=is%3Aopen+is%3Aissue and make sure to include details about which operating system you are running.

matousc89, 2019-01-28:
I am getting the same error, any progress with this?

Running on:
Fresh install of Ubuntu 18.10

I just install and call:

`sudo apt install python3-pip`
`sudo pip3 install muselsl`
`sudo muselsl stream`

at this point a I get the error mentioned above. The same procedure worked on my previous computer with Ubuntu (18.04 I think).

jdpigeon, 2019-01-28:
I was able to replicate this on my system as well (Ubuntu). However, the error only appeared when I attempted to use Muse LSL with `sudo`.

Can you try running without `sudo`? It's likely something about running as root is breaking lib-lsl

matousc89, 2019-01-28:
Thank for quick response!

I tried to reboot and call the `muselsl stream` without `sudo`. Both attempts failled with the same error.

Any other idea?

MuseUK, 2019-01-28:
A couple of points.
1) I am using Python2.7 
2) Does 'muselsl list' show your headset? 
   this return the MAC address of your muse
3) I can then use the 'muselsl stream --address <MAC ADDRESS>' command
4) Normally I get a 'Connecting to Muse <MAC ADDRESS>'
followed by
    No handlers could be found for logger "pygatt.backends.gattool.gattool"
then
    Connection to <MAC ADDRESS> 'failed'

5) This appear to be a timeout message - i I persist, up to 10 times, it does connect and start streaming data.

jdpigeon, 2019-01-28:
Here's what works for me on linux, and a couple of ideas of things to try:

1. Use an Anaconda Python 3 environment
2. `pip install muselsl` (not pip3)
3. `pip install pygatt===3.1.1` (this is necessary to get around a Linux-specific bug in the library's 3.2 version)
4. `sudo setcap 'cap_net_raw,cap_net_admin+eip' `which hcitool` (this should solve Linux issues involving root privilege requirements during scanning)
5. `muselsl stream`

matousc89, 2019-01-30:
Thank for your effort, no success so far. I installed the new version.

The Anaconda approach yield the same result. It does not work, so now I am googling the way how to purge few GB of conda stuff from my system :-)

However I think the conda (or any virtual environment)  should not install anything different than I installed with pip3 (correct me If I am missing something, I would love to learn).

I have the correct version of pygatt. Also I checked the `pylsl==1.13.1`. Is this version the same as you are running?

When I called the muselsl stream, it is running in Python3 really, there should be no problem.

My idea is, that some package is different then it should be - at python or ubuntu level.

1. please can anybody create minimal working environment and freeze it with pip? I would compare the versions of the libs included.

2. does anybody make it working on Ubuntu **18.10**? (As I said, it works for me on old install of 18.04).

jdpigeon, 2019-02-09:
@matousc89 it seems my working conda environment is actually using pylsl 1.10.5. Could you try downgrading to that and see if it fixes your issues?

Hororohoruru, 2019-02-20:
Hey, I arrived to this thread with the same issue and downgrading to pylsl 1.10.5 worked for me. Thanks!

olafurj, 2019-02-20:
confirm downgrading pylsl does the trick.

jdpigeon, 2019-02-20:
Awesome. I'll add this to the common issues section of the FAQ

closed 2019-02-21 by jdpigeon

ErikBjare, 2020-10-26:
Can someone reproduce this with the latest pylsl? I'd like to use a newer version of it and from my initial testing it seems to work fine.

ErikBjare, 2020-11-12:
Just tried the latest pylsl (1.13.6) and didn't get any of the errors mentioned, but I get things like: 

```
Looking for a EEG stream...
Can't find EEG stream.
```

So I assume the issue persists.

xloem, 2022-03-06:
Is this still an issue after two years?

Can anybody describe clear steps to reproduce it?

It's notable that the pylsl version pin is only on linux, so all of the muse-lsl code is still using the latest pylsl on windows and macos.