alexandrebarachant/muse-lsl #79: Add PPG support for Muse 2 devices

jdpigeon, 2019-02-28:
The Muse 2 ships with two additional PPG (photoplethysmography) sensors which can be used to obtain measures of heart rhythm. These sensors broadcast their readings through gatt characteristics in much the same way that the 2016 Muse's gyroscope and accelerometer do.

In order to be able to use PPG with this library, we will have to discover the gatt characteristics of the PPG sensors and include handling functionality to make a PPG stream available for subscription. Code that performs this task for the accelerometer and gyroscope sensors of the Muse 2016 is available in this impending PR: https://github.com/alexandrebarachant/muse-lsl/pull/34. Once a Muse 2 is available for testing, it will likely be most efficient for inclusion of the acc and gyro code to occur alongside development of the new PPG handling code.

kylemath, 2019-03-08:
yes this and #34 would be fantastic, let us know how we can help

jdpigeon, 2019-05-17:
Addressed by https://github.com/alexandrebarachant/muse-lsl/pull/93

closed 2019-05-17 by jdpigeon