alexandrebarachant/muse-lsl #81: Segmentation fault (core dumped) when using muselsl view

Hororohoruru, 2019-03-08:
When already streaming with a muse and calling 'muselsl view' on a different terminal, I get the following error:

Looking for an EEG stream...
Start acquiring data.

                toggle filter : d
                toogle full screen : f
                zoom out : /
                zoom in : *
                increase time scale : -
                decrease time scale : +
               
Segmentation fault (core dumped)

I managed to locally solve the problem before, iirc it had something to do with matplotlib backends.

winsonfzyang, 2019-09-06:
I also have this error when running on Linux Ubuntu.

Hororohoruru, 2019-09-08:
@winsonfzyang try with the optional argument added in #82 , with different matplotlib backends

jdpigeon, 2019-10-22:
This is indeed a problem with matplotlib backends (notoriously finicky between OSes). If anyone runs into seg faults when trying to run the viewer, I recommend trying different backends with the `-b` argument as @Hororohoruru noted.

I've read that the backends are worth trying in this order based on how likely they are to work:

closed 2019-10-22 by jdpigeon