alexandrebarachant/muse-lsl #83: TypeError: descriptor 'encode' requires a 'str' object but received a 'unicode'

ai2010, 2019-03-17:
I am trying to connect with my muse:

muselsl stream --name Muse-NAME
Searching for Muses, this may take up to 10 seconds...
Found device Muse-XXX, MAC Address 00:5XXXX
Traceback (most recent call last):
  File "/home/andrea/.local/bin/muselsl", line 11, in <module>
    sys.exit(main())
  File "/home/andrea/.local/lib/python2.7/site-packages/muselsl/__main__.py", line 52, in main
    cli = CLI(args.command)
  File "/home/andrea/.local/lib/python2.7/site-packages/muselsl/cli.py", line 9, in __init__
    getattr(self, command)()
  File "/home/andrea/.local/lib/python2.7/site-packages/muselsl/cli.py", line 73, in stream
    stream(args.address, args.backend, args.interface, args.name)
  File "/home/andrea/.local/lib/python2.7/site-packages/muselsl/stream.py", line 79, in stream
    "Muse%s" % address,
  File "/home/andrea/.local/lib/python2.7/site-packages/pylsl/pylsl.py", line 193, in __init__
    c_char_p(str.encode(source_id)))
TypeError: descriptor 'encode' requires a 'str' object but received a 'unicode'

Do you have any help?

jdpigeon, 2019-03-18:
I see you're running python 2.7 and I assume you're on Linux? If so, think you've found the weakpoint in [pylsl's support for OS/Python version combinations](https://pypi.org/project/pylsl/) :)

Try upgrading to Python 3.7 and let me know if that fixes the issue. If so, I'll add something to the README indicating that Python 2.7 is necessary on Linux

closed 2019-10-23 by jdpigeon

123alikaha, 2021-03-18:
i have to face a difficulties' run python2.x code in python3.x get an error 
Traceback (most recent call last):
  File "E:/Searchable_Encryption-master/Searchable_Encryption-master/sse_search.py", line 28, in <module>
    keyword_trapdoor = open(keyword_trapdoor).read().strip()
UnicodeDecodeError: 'gbk' codec can't decode byte 0xa5 in position 6: illegal multibyte sequence
Below some part of code that execute in pycharm get an above error 

import pandas as pd
from Crypto.Cipher import AES
from Crypto.Hash import MD5
import time

def build_codeword(ID, trapdoor):
    ID_index = MD5.new()
    ID_index.update(str.encode(ID))
    ECB_cipher = AES.new(trapdoor, AES.MODE_ECB)
    return ECB_cipher.encrypt(ID_index.digest()).encode("hexdigest")

def search_index(document, trapdoor):
    search_result = []
    data_index = pd.read_csv(document)
    data_index = data_index.values
    # start_time = time.time()
    for row in range(data_index.shape[0]):
        if build_codeword(row, trapdoor) in data_index[row]:
            search_result.append(row)

    # print time.time() - start_time
    return search_result

if __name__ == "__main__":

    index_file_name = input("Please input the index file you want to search:  ")
    keyword_trapdoor = input("Please input the file stored the trapdoor you want to search:  ")
    keyword_trapdoor = open(keyword_trapdoor).read().strip()
    search_result = search_index(index_file_name, keyword_trapdoor)
    print("The identifiers of files that contain the keyword are: \n", search_result)