alexandrebarachant/muse-lsl #84: add a new record method to return collected data instead of storing it into disks

thautwarm, 2019-03-18:
I'm developing a real-time application based on muse-lsl, which asks for such an interface with regards of convenience and functionalities: 
```python
def record_to_memory(duration, dejitter=False):
   ...
data = record_to_memory(10)
# do something with `data`
```
Exchanging data with disks heavily hurts the performance.

thautwarm, 2019-03-18:
I've done this in my own package but I think it better to add this functionality into upstream.

```python
from pylsl import StreamInlet, resolve_byprop
from muselsl.constants import LSL_SCAN_TIMEOUT, LSL_CHUNK
__all__ = ['record_stream']

def record_stream():
    """
    :return: a generator to fetch EEG data from existed stream.
    """
    streams = resolve_byprop("type", "EEG", timeout=LSL_SCAN_TIMEOUT)
    if len(streams) == 0:
        raise IOError("Can't find EEG stream.")

    inlet = StreamInlet(streams[0], max_buflen=LSL_CHUNK)

    info = inlet.info()
    description = info.desc()
    Nchan = info.channel_count()

    ch = description.child('channels').first_child()
    ch_names = [ch.child_value("label")]
    for i in range(1, Nchan):
        ch = ch.next_sibling()
        ch_names.append(ch.child_value('label'))

    while True:
        yield inlet.pull_chunk(timeout=1.0, max_samples=LSL_CHUNK)
```

zzxxyy123456, 2024-03-19:
I also want to develop a real-time project,can you share your complete code