alexandrebarachant/muse-lsl #88: Question about muselsl record command

BiaDis, 2019-03-29:
Hi!
I'm working on my dissertation and I'm trying to record some data. I'm using the command "$muselsl record --duration 60" to record and it results in the message of markers not found what I guess it is normal sense I'm not relating to anything else. But after, I lose some data. Wasn't it supposed to be 256 per second? Is it normal to lose some data along the way? (I'm just starting now with the software)

I tried too the command "$muselsl record_direct --duration 60"  but it is not working.

Thank you!

jdpigeon, 2019-04-02:
Hey @BiaDis!

No, it's not normal to lose data while recording. If you open a terminal, run `muselsl stream` and see "started streaming" successfully print, then open another terminal and run `muselsl record` you should get a recording of 256hz EEG.

If you're running into issues could you paste the error messages you are seeing here?

jdpigeon, 2019-04-04:
That sounds totally reasonable actually. I wouldn't be surprised if there was a little bit of starting and ending lag in the recording process that might make you seem to lose a few samples. Also, the nature of bluetooth streaming is such that sampling rates will seem to vary. Hopefully this small sample rate weirdness won't affect your work.

closed 2019-04-04 by jdpigeon

BiaDis, 2019-04-08:
Thank you! I'm trying to make an accurate FFT of the data. Another thing, the data is lost just in the starting and ending, never during right? Because if it is accuracy it will be a problem...

jdpigeon, 2019-04-08:
An increased risk of timing issues is a tradeoff that you make when working with bluetooth-based EEG, but at least on the Linux OS I haven't had any issues in Muse/Muse LSL data quality and have been able to replicate some [classic EEG findings](https://github.com/NeuroTechX/eeg-notebooks)

If you're interested, I think this paper still does the best job of describing the pros and cons of Muse research: https://www.frontiersin.org/articles/10.3389/fnins.2017.00109/full