alexandrebarachant/muse-lsl #92: Muse LSL not detecting marker stream from PsychoPy

jctaylormru, 2019-05-12:
This is undoubtedly my error, but I can't find any solution online after a couple of hours of searching.  I'm trying to stream both muse input and psychopy (stimulus presentation program) marker input and record into the same file.

I can record from the Muse to a file using BlueMuse to establish the connection and `>muselsl record --duration 5` in an Anaconda window

I am using test code from https://github.com/kaczmarj/psychopy-lsl to try to create a marker stream.  I have:

1. Tried running example_coder.py from within the Coder of Psychopy after starting muselsl record in an Anaconda window - I get a "ModuleNotFoundError" for pylsl in the Coder (despite having installed it through an Anaconda prompt)
2. Tried running example_coder.py from the Anaconda window using `>python example_coder.py`.  The program runs, but no marker stream shows up in the LSL Bridge and no marker stream shows up in the saved .csv file.  This is when I run stimulus display program in a different Anaconda window than the muselsl record program.
3. Tried running both programs (muselsl and example_coder.py) from the same Anaconda window using `>python psychopy-lsl-master\example_coder.py & muselsl record --duration 5` and the programs run sequentially (no matter what order I put the programs in) instead of in parallel.  

So I'm stumped.  I assume I need to run the programs in different windows, but can't get the marker stream to be recognized in that case.  I'm doing something wrong.  Please tell me what!

Many thanks

Jamie

jdpigeon, 2019-05-16:
Your second approach is the right one. The first reason I can think of that it wouldn't work is that you ran the`muselsl record` command  significantly earlier than starting the stimulus presentation. `muselsl record` will wait for a couple seconds as it looks for an LSL Markers stream before recording EEG.

Have you tried running the stimulus presentation first or at least trying to run both commands as simultaneously as possible?

If you're more comfortable with Jupyter Notebooks, there's also the [EEG Notebooks](https://github.com/NeuroTechX/eeg-notebooks) project which demonstrates how to run psychopy experiments with MuseLSL and has some stimulus presentation programs you might be able to adapt to your purposes.

jctaylormru, 2019-05-16:
Thank you very much for your reply!

I ran the program (`example_coder.py`) that creates the marker stream (which takes something like 20 seconds to finish, sending markers on 1 second intervals, I believe) very shortly (< 1 sec) before issuing `>muselsl record` command in the second anaconda window.   I don't think it was too short an interval between, but I'll comment on that below.

I have some programming background, but have never worked with Jupyter notebooks.  I can learn those, as a last resort, but will try to use my existing skills first (so much to learn, so little time!).  I need to be able to program novel experiments (using PsychoPy) so I feel like this would be a steep learning curve.

However, you have given me some ideas to try first.

One possibility is that I need the marker stream program to be up and running further in advance of the `>muselsl record` command.  Perhaps there's some lag in the marker program that's causing problems.  I don't actually think so (having read through the code), but it can't hurt to try.

Another possibility is that I need to start the marker stream before I start up BlueMuse, so that the LSL Bridge picks up the marker stream when it (the bridge) starts.  I don't know if BlueMuse continuously looks for new streams, so perhaps that's the problem.

Finally, I can do some looking at code examples from the EEG Notebooks to see if there is some difference between what is happening in them and what is happening in `example_coder.py`.

Thank you for the help so far - I'll post a solution if I find one, and hope you'll be okay with me continuing the query if I don't.

jdpigeon, 2019-05-16:
Sorry to hear that's it not working when you run the two commands at (nearly) the same time. That indicates there's something going wrong here, probably on the LSL side.

Don't worry about BlueMuse, that shouldn't interact with your markers stream. What BlueMuse does is establish your EEG stream. That seems to be working because you're getting EEG data saved.

Does the terminal running `muselsl record` say `Can't find Markers stream.`? If it does then there's either an issue with your stimulus presentation being unable to establish a markers stream or pylsl being unable to find it.

I think looking into EEG notebooks is a good idea too. If you're not able to get those to work then I would suspect there's something wrong with your python environment. It would be good to figure that out.

jctaylormru, 2019-05-16:
Success!  Sort of.

It wasn't the timing.  It was a code thing.  The original code in example_coder.py used
```
  info = StreamInfo(name='example_stream', type='Markers', channel_count=1,
                      channel_format='int32', source_id='example_stream_001')
```
but the code in one of the eegnotebooks (go_nogo.py) used
  `info = StreamInfo('Markers', 'Markers', 1, 0, 'int32', 'myuidw43536')`
When I replaced the former with the latter, the marker stream appeared.  Though oddly it was not called "Markers" but "Marker0".  I tried changing the first string to "MS2" but that doesn't work (the stream is no longer found), even though the first field should be the name= field (I think, based on the StreamInfo definition in pylsl).   So something is up with the setting of a name, but I don't have time to chase it down. 

One other observation - it's necessary to include a timestamp in order for the markers to end up in the right spot.  The original code used
   `outlet.push_sample(markers['world'])`
which only put a single marker at the top of the output file.  Using
```
     timestamp = time()
     outlet.push_sample(markers['world'],timestamp)

```
inserted the markers at (from what I can tell, based on their temporal separation) correct spots.  I'm slightly confused about the necessity of this, as I thought the lab streaming layer was supposed to work across machines (which might have different system times) but, as above, I only need it to work on one machine, so I am content.

jdpigeon - thank you very much for pointing me in the direction of the notebooks.  I can cobble together a working solution now!

closed 2021-01-10 by jdpigeon