alexandrebarachant/muse-lsl #94: PPG Recording or viewing is not working.

iPsych, 2019-05-20:
Start the streaming works fine.
`$ muselsl stream --ppg
Searching for Muses, this may take up to 10 seconds...
Found device Muse-XXXX, MAC Address 00:55:DA:B5:06:XX
Connecting to Muse-XXXX : 00:55:DA:B5:06:XX...
Connected.
Streaming EEG PPG...`

But whenever try to record PPG, it cannot find the stream.

`record --type PPG
Looking for a PPG stream...
Can't find PPG stream.
`

--type EEG
just works fine.
`record --type EEG
Looking for a EEG stream...
Started acquiring data.
Looking for a Markers stream...
Can't find Markers stream.
Start recording at time t=1558332757.922
Time correction:  -1.2806500308215618e-05
Time correction:  -1.7711998225422576e-05
Done `

iPsych, 2019-05-20:
I changed slightly csl, view, viewer_2 following stepheninx, but still cannot find PPG stream.
It seems broadcasting shows 'Streaming EEG PPG...' but not actually streaming PPG.

jdpigeon, 2019-05-20:
Can you report which OS and which version of Python you're using.

Also, could you trying adding the `--disable-eeg` flag to `muselsl stream --ppg` and see if you are able to record from a PPG stream then?

iPsych, 2019-05-22:
@jdpigeon 
Hi, I just tested and confirmed it works when use --disable-eeg. but when the EEG stream starts together, muselsl cannot find PPG stream.

closed 2019-10-23 by jdpigeon