alexandrebarachant/muse-lsl #95: Error on Microsoft Windows [Version 10.0.16299.1087] - Bluemuse 1.1.0.0

sorenopsy, 2019-05-20:
Hello this what i get when i try to start muselsl last version with BlueMuse 1.1.0.0
BlueMuse recognize device and stream correctly.
Python is 2.7.
Windows version is:
Microsoft Windows [Version 10.0.16299.1087]
Regards,
Giampaolo
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
[image]
[image]

  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "c:\python27\scripts\muselsl.exe\__main__.py", line 5, in <module>
  File "c:\python27\lib\site-packages\muselsl\__init__.py", line 1, in <module>
    from .stream import stream, list_muses
  File "c:\python27\lib\site-packages\muselsl\stream.py", line 162
    print(f'Streaming{" EEG" if not eeg_disabled else ""}{" PPG" if ppg_enabled else ""}{" ACC" if acc_enabled else ""}{" GYRO" if gyro_enabled else ""}...')

jdpigeon, 2019-05-20:
Thanks for reporting this. Will upload a fix soon

jdpigeon, 2019-05-20:
Can you try upgrading to Python 3 and report if this issues still appears?

hyruuk, 2019-05-21:
I have a similar issue on Ubuntu 18.04 with Python 3.5 : 
`(cbc) hyruuk@hyruuk-yoga:~$ muselsl list
Traceback (most recent call last):
  File "/home/hyruuk/anaconda3/envs/cbc/bin/muselsl", line 7, in <module>
    from muselsl.__main__ import main
  File "/home/hyruuk/anaconda3/envs/cbc/lib/python3.5/site-packages/muselsl/__init__.py", line 1, in <module>
    from .stream import stream, list_muses
  File "/home/hyruuk/anaconda3/envs/cbc/lib/python3.5/site-packages/muselsl/stream.py", line 162
    print(f'Streaming{" EEG" if not eeg_disabled else ""}{" PPG" if ppg_enabled else ""}{" ACC" if acc_enabled else ""}{" GYRO" if gyro_enabled else ""}...')
                                                                                                                                                           ^
SyntaxError: invalid syntax
`

sorenopsy, 2019-05-21:
> Can you try upgrading to Python 3 and report if this issues still appears?
[image]

On python 3.7 seems to works fine, bluemuse is started and device is recognized.
Problem is that when i use python -m muselsl view the graph crash abnormally.
The record command instead works fine:
C:\python -m muselsl record -d 5 -f c:\Users\xxx\Desktop\test.txt
Looking for a EEG stream...
Started acquiring data.
Looking for a Markers stream...
Can't find Markers stream.
Start recording at time t=1558431832.577
Time correction:  -2.6879992219619453e-05
Time correction:  -1.7705999198369682e-05
Done - wrote file: c:\Users\xxx\Desktop\test.txt.

C:\more c:\Users\xxx\Desktop\test.txt
timestamps,TP9,AF7,AF8,TP10,Right AUX
1558431833.226,-857.422,-672.363,-678.711,-943.359,-1000.000
1558431833.230,719.238,-1000.000,-1000.000,-999.023,-174.805
1558431833.234,-670.410,-46.875,-99.121,-213.867,-963.867
1558431833.238,-1000.000,-876.465,-885.254,-1000.000,971.680
1558431833.242,-164.551,833.984,852.539,-253.906,-24.414

sorenopsy, 2019-05-21:
closed by mistake sorry :)

sorenopsy, 2019-05-21:
Update:After installing pyqt5 it works with view version 2.
[image]

jdpigeon, 2019-05-21:
I used a new python 3.6 feature, [f-Strings](https://realpython.com/python-f-strings/) (really nice to work with_ in version 2.0 that's causing this issue.

When I have some time I can downgrade to a Python <3.6 compatible implementation of our string construction functions. Until then, this issues should be resolvable by using Python > 3.6

nrcmichael, 2019-06-13:
@sorenopsy can you confirm which version of the Muse headband you are using?

I'm trying to get the Muse 2 working with something in Windows and no luck so far. After working through threads on the Muse website and for Bluemuse and here I don't think there is any confirmation of anybody getting it to work with Windows.

closed 2019-10-23 by jdpigeon