alexandrebarachant/muse-lsl #97: Citing muse-lsl in papers

thautwarm, 2019-05-21:
As muse-lsl has become a super useful tool in muse device based EEG data analysis, it seems there're getting more and more papers to cite it, so will there be a guide for citing items?

alexandrebarachant, 2019-05-25:
Thanks for reporting this. I created a DOI for the package, and will update the README once I have the correct citation. We might write a small paper down the road ...

alexandrebarachant, 2019-05-25:
here is the citation :

```
@misc{muse-lsl,
  author       = {Alexandre Barachant and
                  Dano Morrison and
                  Hubert Banville and
                  Jason Kowaleski and
                  Uri Shaked and
                  Sylvain Chevallier and
                  Juan Jesús Torre Tresols},
  title        = {muse-lsl},
  month        = may,
  year         = 2019,
  doi          = {10.5281/zenodo.3228861},
  url          = {https://doi.org/10.5281/zenodo.3228861}
}
```

> Alexandre Barachant, Dano Morrison, Hubert Banville, Jason Kowaleski, Uri Shaked, Sylvain Chevallier, & Juan Jesús Torre Tresols. (2019, May 25). muse-lsl (Version v2.0.2). Zenodo. http://doi.org/10.5281/zenodo.3228861

@jdpigeon @hubertjbanville @kowalej @urish @sylvchev @Hororohoruru

closed 2019-05-25 by alexandrebarachant