alexandrebarachant/muse-lsl #98: Headband contact status (horseshoe)

mightreya, 2019-05-23:
Hi,
Is there a way status indicator data for each channel, to have an idea if Muse has a proper contact?
In case it's not implemented yet, could you please point out how to do that (guess it's about a correct constant), can try to contribute.

jdpigeon, 2019-05-23:
Hey there! We don't have a signal quality metric in Muse LSL yet. Usually we just quickly open up the EEG viewer to look at signal quality before running an experiment. In viewer V1 (`muselsl view`), there should be numbers next to each channel label. These represent the standard deviation of the signal in the window and, in our experience, should be below 20 in order to get good results.

closed 2019-05-25 by jdpigeon

oori, 2019-12-12:
@jdpigeon How would you recommend to mimic `touching_forehead` ?
Back in the "muse SDK" days, it used to work quite well (`/muse/elements/touching_forehead`).    It wasn't perfect, but good enough for "game on" ux.   I'm not sure, but assume they did it via the DRL sensor at Fpz (reference sensor).

We can't read the reference sensors via muse-lsl, or can we?     otherwise,  settle on assumptions (e.g. `touching_forehead ~=  std < 50 on all channel, for 1-2 seconds`)

thanks

jdpigeon, 2019-12-12:
You're right, the reference sensor isn't accessible through bluetooth. The best thing we can do would be some metric like you've proposed based on standard deviation