alexandrebarachant/muse-lsl #99: using the right marker timestamp from matlab

HashemiScience, 2019-05-24:
tl;dr: How can I use lsl_local_clock to send precise markers from matlab to muselsl via outlet.push_sample(marker,timestamp)?

Problem:
When sending markers in matlab using outlet.push_sample(marker, timestamp), using timestamp of 0 or leaving blank does not use the current lsl_local_clock (despite what documentation suggests). It actually will just send the flag to the very first sample in the dataset. So i’m forced to either:

1. Use a high precision time from matlab/psychtoolbox. This works, but the clocks between matlab and muselsl seem to accumulate a delay between them as a function of muselsl stream runtime duration. See attached figure to show how the delay between GetSecs in matlab and the final timestamp from [~, lsltime] = inlet.pull_chunk grows as a function of time (x-axis = sample #, y-axis = difference between GetSecs and lsltime.). I collected this over a 2 hour period, sampling GetSecs and lsltime every 1 second.
[image]
zoomed in to see each sample over a 5 minute time period:
[image]

2. Before sending a marker from matlab, use inlet.pull_chunk and use the latest timestamp to send the flag. This avoids the accumulating delay, but is too jittered. I can’t use pull_sample because that just pulls the next sample available in the stream, not the most recent sample.

apparatus:
- ubuntu 16.04
- muselsl v1.0.0
- matlab 2018a with Psychtoolbox 3.0.15 and labstreaminglayer
- muse 2016

setup to collect data for the above figures to show the accumulating delay...
open muselsl stream
open muselsl record
in matlab, ran:
```
lib = lsl_loadlib();
info = lsl_streaminfo(lib,'Marker','Markers',3,0,'cf_string');
outlet = lsl_outlet(info);

disp('Resolving an EEG stream...');
result = {};
while isempty(result)
    result = lsl_resolve_byprop(lib,'type','EEG');
end

% create a new inlet
disp('Opening an inlet...');
inlet = lsl_inlet(result{1}); % create inlet for getting proper timestamps

% collect data for 2 hours, sending a flag every 1 second... (change 7200secs / 60 = 120 mins)
tEnd=GetSecs;
for ii = 1:7200
    
    fprintf('Second %i of 7200... \n', ii);
    
    while(GetSecs-tEnd) < 1
    end
    tEnd = GetSecs;
    
    [~, lsltime] = inlet.pull_chunk;
    tcor=GetSecs-tEnd;
    if ~isempty(lsltime)
        lsltime = lsltime(end);
    else
        [~, lsltime] = inlet.pull_sample;
        tcor=GetSecs-tEnd;
    end
    outlet.push_sample({'flag', num2str(tEnd), num2str(tcor)}, lsltime-tcor)
    
end

```

The accumulating delay is slow and is linear. We discovered this delay when collecting actual data for an ERP study. Here is the single-trial ERP at time 0 and again, for the same participant and experiment, 2 hours later. The delay accumulates slow enough that you can get an ERP in a 5 minute task, but you cannot reasonably combined trials collected with a significant delay (scale of several minutes) between them. In these plots, the ERP is time-locked to the time of stimulus presentation (obtained via high-precision VBLTimestamp of the output of Screen('Flip') in matlab/psychtoolbox). You can even see the accumulation of the delay in the single-trial ERPs... 
[image]
[image]
I can time-lock these to the LSLtime obtained via inlet.pull_chunk, and it _sorta works_, but it's way too jittered to be a good solution.

Some other things that might be worth noting: I've tried turning the dejitter in muselsl off, and it doesnt seem to make any difference. The delay is entirely dependent on when the muselsl stream has started...one temporary solution is to restart the stream every few minutes to minimuze the effects of the delay, but that's a really bad solution. The delay is seen whether muselsl and matlab are on the same or different computers. I only have access to linux computers, so maybe it's OS dependent? It is only on linux that the system clock that LSL and Psychtoolbox/matlab use are the same.

jdpigeon, 2019-05-24:
Ali, thank you for the very clear and figure-supported issue.

First, let me just confirm what Muse LSL does. Every time a packet comes in it uses the python `time` function to generate a timestamp of the most recent sample, then automatically derives the timestamps for all previous samples in the packet based on index and sampling rate. We've had some timestamp issues due to Bluetooth jitter in the past, but I don't see how our implementation could lead to a linear accumulating discrepancy.

That likely doesn't even matter though because you're seeing delay on Matlab's side, that's nuts! The fact that the Matlab LSL API seems to not be working the way you intend is a bad sign too.

Have you only noticed this issue with Matlab while also running Muse LSL? For an experiment, maybe you could you write a Python version of your script that connects to your Matlab LSL stream and compares it's timestamps to python's `time` function?

alexandrebarachant, 2019-05-25:
Hey, 
this is a classic case of clock drift.

the timestamps on muse-lsl are based on the indices of the sample and the sampling frequency of the signal see [here](https://github.com/alexandrebarachant/muse-lsl/blob/master/muselsl/muse.py#L332). Because of the tolerance of electronic component, the sampling frequency is never an exact 256Hz, but rather a 256.0001 or something, which will create a linearly increasing error on the timestamps.
one alternative is to affect timestamp the time they are received by the laptop, but because of bluethoot jitter, this is not accurate enough. 
the optimal solution is to apply a timestamp correction to compensate for this. We actually have the [architecture](https://github.com/alexandrebarachant/muse-lsl/blob/master/muselsl/muse.py#L299) for that but it has never been implemented (this is a hard problem).

However, the record function of muse-lsl has an offline dejittering option (see [here](https://github.com/alexandrebarachant/muse-lsl/blob/master/muselsl/record.py#L87)) that will account for this by doing a linear interpolation of the timestamp when the recording is done.

here are some different option you can use:
- save both the timestamp you receive the packet and the timestamp provided by LSL, and apply a correction like we do in record.
- or, do smaller recordings. this wont leave enough time to build a strong drift. the downside is that you might get a slightly different offset across recording.
- or make a PR to finish the implementation of the drift correction in muse-lsl directly (could be as simple as estimating the true sampling frequency online by counting the number of sample received in a given period of time.)

alexandrebarachant, 2019-05-25:
Okay, i made a PR that should fix the issue by correcting the timestamps directly in the muse base class. 

Looking more closely at the matlab code you provided, i think the issues come from the fact that you are trying to apply a correction to your marker timestamps before sending them.

if you are using muselsl to record you data, you can just send the markers with the timestamp corresponding to you local machine time, and let the recorder do the correction i.e. no need to pull emg sample in matlab to get their timestamp.

HashemiScience, 2019-05-27:
Wow, you guys are awesome!! I didn't expect such a quick fix.

The correction (tcor) i applied is negligible, and was just one of my (unsuccessful) attempts to reduce some jitter. I will revert to using the system time once I update my muselsl!

HashemiScience, 2019-05-27:
@alexandrebarachant your adaptive timestamp works perfectly. I just collected data on the same experiment (3 minutes of data collection), 2 hours apart, and got these beautiful ERPs without any sign of the drift.

THANK YOU!

for the record, I am sending my markers using the high precision time of the local computer (GetSecs in matlab).

Time 1:
[image]

Two hours later:
[image]

closed 2019-05-27 by HashemiScience

alexandrebarachant, 2019-05-28:
I'm glad this worked for you.
I probable have to refactor a bit more code to do the same thing for other sensors modality, then i will merge this.

PR #101 (2021-02-14): Adaptive timestamp correction
This PR implement the adaptive timestamp correction described in https://arxiv.org/pdf/1308.3846.pdf (i used the least square estimator).

I need to perform further testing, but it should fix #99 
@HashemiScience can you git it a try ?