alexandrebarachant/muse-lsl #100: Error with muse 2016 from latest version

JohnGriffiths, 2019-05-25:
It looks like the latest changes to accomodate the additional muse 2 sensor has broken muselsl for the muse 2016. 

We encountered this several times in the past week on windows machines. 

I don't have the exact error to hand right now, but 

1. It occurred when running `muselsl view` for first time, after initiating a stream with bluemuse
2. It mentioned something like 'PPG' and array sizes, and so I guessed was from the muse 2 updates
3. Temporary fix for this was 

```bash
pip uninstall muselsl  
git clone https://github.com/alexandrebarachant/muselsl  
cd muselsl  
git checkout to an arbitrarily chosen earlier commit: 43d13c300ec885cfacd327d26952388edc09eb2f
python setup.py install   
```

jdpigeon, 2019-05-25:
Hey Jon, sorry to hear this.

I just confirmed that Muse 2016 still works on version 2.0.1 on Linux, as long as you don't try and request a PPG stream.

I'm not sure where the error you noticed would be arriving, but it's likely due to interaction with BlueMuse. Could you confirm that the users haven't been trying to request a PPG stream from their Muse 2016's and perhaps post the stack trace of the error you're seeing when you have a chance?

JohnGriffiths, 2019-05-25:
Just called `muselsl view`. So it's whatever the default behaviour is.

kowalej, 2019-05-26:
I should probably add the new data channels in BlueMuse for the Muse 2.

JohnGriffiths, 2019-05-26:
Sure Jason do an update to bluemuse. But the fix to this problem shouldn't be at bluemuse it should be at muse-lsl accomodating input stream whatever shape / version it's coming from. Otherwise there will be immediately be a bunch of backwards compatibility issues with prev bluemuse versions and I don't like the idea of immediately forcing a bluemuse update to everyone using EEG notebooks. There's already a lot of user energy+time+patience going into getting successful bluemuse connections, and a certain amount of random luck as to whether that's successful. Need to minimize sources of variability there.

kowalej, 2019-05-26:
This would not be a fix for muselsl but an enhancement. Like you said we should maintain backwards compatibility.

jdpigeon, 2019-05-27:
I think there's a couple things going on here that we should track separately.

1. `muselsl view` breaking on Muse 2016 devices is strange because this library's viewers weren't modified in the most recent update and, AFAIK, BlueMuse wasn't updated as well. If you see this error again, @JohnGriffiths, can you create another issue with the full error message so we can get a better idea of what's going on? I think it's possible that you may have encountered some python version incompatibility issues that I fixed in a recent patch.

2. BlueMuse updates to add PPG, ACC, and GYRO support. This would certainly be nice to have, but shouldn't be a blocking anybody unless they really want to run PPG experiments on Windows. Completion is of course dependent on Jason's time and acquisition of a Muse 2. Data flow between muse-lsl and BlueMuse is purely through LSL, so it's relatively easy to maintain backward compatibility.

closed 2019-05-27 by jdpigeon