alexandrebarachant/muse-lsl #104: possible error in examples/utils.py

Dutchflag, 2019-08-30:
i've been playing around with the neurofeedback example, and i noticed the following.

In line 85 of utils.py, the PSD is calculated. However, without taking the square of the magnitude of the FFT, you get the true theoretical FFT and not the power spectrum. I think the **2 is missing?

jdpigeon, 2019-10-23:
Good catch. I think we can update that function soon to provide a true Welch's PSD implementation through scipy.

oori, 2019-11-17:
> Welch's PSD implementation 

@jdpigeon, For reference code and explanation see [Compute the average bandpower of an EEG signal](https://raphaelvallat.com/bandpower.html) or his YASA library's [yasa.bandpower](https://raphaelvallat.com/yasa/build/html/generated/yasa.bandpower.html)  function