alexandrebarachant/muse-lsl #107: Where is muse-lsl.py?

jscodedeep, 2019-10-01:
What's in the muse-lsl.py?

jdpigeon, 2019-10-02:
`muse-lsl.py` was the filename used in the library before we updated it to be a CLI. If you're seeing that somewhere, please let me know and I'll remove the reference

jscodedeep, 2019-10-03:
It's from some of the Issues.

Could not install BlueMuse because the certificate has expired, bought BLE112 dongle with Windows 10, tried following commands, all failed or asking BlueMuse, I can see COM4 in the DeviceManager

>muselsl stream -b bgapi
Starting BlueMuse, see BlueMuse window for interactive list of devices.

>muselsl stream -i COM4 -b bgapi
Starting BlueMuse, see BlueMuse window for interactive list of devices.

>muselsl stream -a 00:55:DA:B5:38:CB -b bgapi
Connecting to Muse: 00:55:DA:B5:38:CB...
Connection to 00:55:DA:B5:38:CB failed

closed 2019-10-22 by jdpigeon