alexandrebarachant/muse-lsl #110: Bluetooth disconnecting when PPG streaming (Muse_2 issue)

tomek-, 2019-11-26:
It seems that Bluetooth is disconnecting when PPG streaming is on after about 10~15 minutes for Muse_2. No problem at all (30+ minutes testes) when any other combination without PPG is set. Any buffering issue? Any help as having pulse information would be nice for ML!

jdpigeon, 2019-11-26:
Are there any error messages printed when the bluetooth disconnects? I'm curious if this is a device issue (the Muse wasn't really built for sessions longer than 30 minutes)

oori, 2019-12-01:
I can partially confirm this, at least enough to keep this issue afloat..
I see frequent disconnection using muse 2018, with or without PPG.   No special errors printed other the the usual `missing sample .... : ....`, before it `Disconnected.`

On same setup the Muse 2016 works **very** stable!
I'm using MacOS+BLE112.   I need to test on another stack, Windows machine w/ Bluemuse.

tomek-, 2019-12-03:
Right, I do confirm @oori message and I have the same problem on MacOS+BLE112 and Win10+BLE112 :( We have several BLE112 and my collaborators could reproduce the problem with several Muse2018 headsets.

tomek-, 2019-12-03:
@jdpigeon - right, but from my experience running 2 hours long streaming sessions with MUSE_2016 is completely no problem with the fabulous **muse-lsl** software ;)

tomek-, 2019-12-03:
Below is an example with 6 minutes long streaming (see time stamps on the right) with a suddenly broken connection using MUSE_2:

[image]

oori, 2019-12-15:
I've tested with higher number (`10`, `60`) for [AUTO_DISCONNECT_DELAY](https://github.com/alexandrebarachant/muse-lsl/blob/520cac35d74993b9dd1c6d42a73f484f6b2d906a/muselsl/constants.py#L34).  I did get a long 42min session, but on the other hand, a bunch of short ones as well (_9m, etc.._). Does **not** seem to be the issue.

@jdpigeon Any tip how to debug this ?            it's a consistent issue, and makes the muse 2018 unusable for some user applications.

See also: https://github.com/alexandrebarachant/muse-lsl/issues/55 and the linked issues.

oori, 2019-12-16:
The previous tests were on MacOS.    Now on Raspbian (on rPi 4) with internal bluetooth, we get very clear results:
`EGG + ACC + PPG`   immediately gives `missing sample` (_10-20 times_) and then disconnects in a couple of seconds.
`EEG + PPG` _same_
`EEG + ACC`  starts giving `missing sample` errors only after half a ~30 seconds, then disconnects after another ~30 seconds.   Overall - x10 times longer than the unusable PPG.
`EEG` works **very good**. long sessions.   might throw a `missing sample` or two, but doesn't disconnect !
Note: while working, it **does** stream data !

So, I read [this issue](https://github.com/peplin/pygatt/issues/187), and test the RPi with BLED112 and BGAPI backend, and the results are extremely better, similar to MacOS.   
`EGG + ACC + PPG` long steady 30+ minutes sessions, but also short ones.

So, with BGAPI, we can get long sessions, but it's not promised.   Software restarts can be fast (using saved `address`), the only problem I see now is that _sometimes_ after `Disconnect`, the muse is "stuck", _as if_ it's still connected, and doesn't jump back to "discovery mode".   So, one has to take off the headset and restart.  But luckily, using BGAPI, I see this phenomena much less often as with GATT.

5anirban9, 2021-08-13:
I am having the same problem. The EEG stream is getting disconnected suddenly without giving any error message while I was trying to receive it through lsl using a python script. Any solution for that?

buyuk-dev, 2021-08-15:
@5anirban9 Have you tried increasing the AUTO_DISCONNECT_DELAY?

mohammad-rahmani, 2022-05-06:
Same experience here with my [Ubuntu laptop or Raspberry Pi 4B] + Muse2018 device. Stream ends after some random time ranging from a few seconds (rarely) to around 90 minutes (with somewhere about an hour being the most frequent case).

Also tried a Nordic PCA10059 bluetooth dongle but faced the same issue. 
By any chance, is there a clear understanding of the reason for this by now? (Or probably a workaround by any chance, of course).