alexandrebarachant/muse-lsl #111: muselsl view: Segmentation fault (Core dumped)

amar-enkhbat, 2019-12-05:
Hello, like many others I'm working on a real-time eeg application. 
Just started using muse-lsl and muse stream is working fine. However, when I run muse view following error occurs. Any idea how to fix this?
 Looking for an EEG stream...
`Start acquiring data.

                toggle filter : d
                toogle full screen : f
                zoom out : /
                zoom in : *
                increase time scale : -
                decrease time scale : +
               
Segmentation fault`

oori, 2019-12-08:
Try using v2:  `muselsl view --version 2`

amar-enkhbat, 2019-12-09:
Version 2 isn't working either. I'll post error report later.

llamasking, 2019-12-17:
I have this issue, too. A workaround I found to work is to use the Qt5Agg backend with `-b Qt5Agg`. You'll need to install the PyQt5 package with pip to use it, though.

amar-enkhbat, 2019-12-17:
> I have this issue, too. A workaround I found to work is to use the Qt5Agg backend with `-b Qt5Agg`. You'll need to install the PyQt5 package with pip to use it, though.

Thanks. Works with the Qt5 module.

amar-enkhbat, 2019-12-17:
> Try using v2: `muselsl view --version 2`

Version 2 didn't work previously but now works fine after intalling qt5 lol

obo, 2020-12-10:
I am absolutely new to EEG, I managed to record raw signal with ``muselsl record --duration 60`` but I am failing to get any of the viewers running. Both version 1 and version 2 say this:

```
(venv2) 09:41 obonb experiment$muselsl view  -b Qt5Agg
Looking for an EEG stream...
Start acquiring data.
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)
```

Given the multitude of possible graphics backends, could you recommend the one with the least dependencies and generally easiest to install? I tried e.g. Gtk3Agg but it requires cairo and installing cairo fails for me (Requested 'cairo >= 1.15.10' but version of cairo is 1.14.6) because I still use Ubuntu 16.04.

obo, 2020-12-10:
Trying out the various possible backend, the furthest I got was this.

```
sudo apt-get install python3.7-tk
```

And then the viewer runs of half a second:

```
(venv2) 09:56 obonb experiment$muselsl view  -b TkAgg
Looking for an EEG stream...
Start acquiring data.

                toggle filter : d
                toogle full screen : f
                zoom out : /
                zoom in : *
                increase time scale : -
                decrease time scale : +
               
Segmentation fault (core dumped)
```

The plot appears, with most signals flat zero and possibly the very last part showing some data, but then it immediately crashes with a core dump. Any ideas?

obo, 2020-12-10:
Downgrading pycairo (``pip install pycairo==1.19``), I can also start TkCairo backend (only in viewer version 1, version 2 has the qt error  ``Could not load the Qt platform plugin "xcb"``).

Unfortunately, TkCairo crashes with non-deterministic error messages. The window with charts appears for half a second always:

### Attempt 1

```
(venv2) 10:02 obonb experiment$muselsl view  -b TkCairo
Looking for an EEG stream...
Start acquiring data.

                toggle filter : d
                toogle full screen : f
                zoom out : /
                zoom in : *
                increase time scale : -
                decrease time scale : +
               
Segmentation fault (core dumped)
```

### Attempt 2

```
(venv2) 10:02 obonb experiment$muselsl view  -b TkCairo
Looking for an EEG stream...
Start acquiring data.

                toggle filter : d
                toogle full screen : f
                zoom out : /
                zoom in : *
                increase time scale : -
                decrease time scale : +
               
python3.7: ../../../../src/cairo-scaled-font.c:1317: cairo_scaled_font_destroy: Assertion `! scaled_font->cache_frozen' failed.
Aborted (core dumped)
```

### Attempt 3

```
(venv2) 10:03 obonb experiment$muselsl view  -b TkCairo
Looking for an EEG stream...
Start acquiring data.

                toggle filter : d
                toogle full screen : f
                zoom out : /
                zoom in : *
                increase time scale : -
                decrease time scale : +
               
Exception in Tkinter callback
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/muselsl/viewer_v1.py", line 150, in update_plot
    self.fig.canvas.draw()
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/backends/backend_tkcairo.py", line 21, in draw
    self.figure.draw(self._renderer)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/artist.py", line 41, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/figure.py", line 1864, in draw
    renderer, self, artists, self.suppressComposite)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
    a.draw(renderer)
[... 34 lines cut ...]
    return draw(artist, renderer, *args, **kwargs)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/cbook/deprecation.py", line 411, in wrapper
    return func(*inner_args, **inner_kwargs)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/axes/_base.py", line 2747, in draw
    mimage._draw_list_compositing_images(renderer, self, artists)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
    a.draw(renderer)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/artist.py", line 41, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/axis.py", line 1169, in draw
    tick.draw(renderer)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/artist.py", line 41, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/axis.py", line 291, in draw
    artist.draw(renderer)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/artist.py", line 41, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/lines.py", line 763, in draw
    gc = renderer.new_gc()
  File "/home/bojar/diplomka/granty/expro-neurem3/neurem3-eyetracking/experiment/venv2/lib/python3.7/site-packages/matplotlib/backends/backend_cairo.py", line 311, in new_gc
    self.gc.ctx.save()
cairo.Error: Context.restore() without matching Context.save()
```

Based on Attempt 2, could all these issues be caused by some fonts issues, e.g. when the viewer tries to add some label (tick labels, legend, bolded something, whatever?)

xloem, 2021-01-07:
I wasn't aware of `--version 2` and resolved this locally by rebuilding `tcl`, `tk`, and `matplotlib` from source.

jdpigeon, 2021-01-10:
> Given the multitude of possible graphics backends, could you recommend the one with the least dependencies and generally easiest to install? I tried e.g. Gtk3Agg but it requires cairo and installing [...]

The matplotlib backend we've found usually works the best is TkAgg, which is provided as the default argument to the `backend` parameter. However, we recognize that matplotlib backends can be finicky depending on your environment, so we also provide viewer v2 (which uses vispy instead of matplotlib) to hopefully get around those issues.

rmib200, 2021-02-01:
Are there any fixes for the V1? I'm able to stream the V2 but the font sizes are incredibly small (dot size). I want to chech that the variability of my signals are under 10. Is there a way around?
Capture of the v2 with the impossible to see text.
[image]

jdpigeon, 2021-02-01:
Unfortunate about the small text size. In a pinch, I'd say you could fix that by changing your screen's display scaling if you're on a high resolution screen. Otherwise, I can confirm that the text just lists the channel name on the left and the variance over the displayed window on the right.