alexandrebarachant/muse-lsl #114: TypeError by starting stream

b3n3w, 2019-12-28:
I'm getting this error message below when starting a normal lsl stream. Already tried to update python to 3.7 and changed the pygatt version.

Traceback (most recent call last):
  File "/home/usability/.local/bin/muselsl", line 10, in <module>
    sys.exit(main())
  File "/home/usability/.local/lib/python2.7/site-packages/muselsl/__main__.py", line 57, in main
    cli = CLI(args.command)
  File "/home/usability/.local/lib/python2.7/site-packages/muselsl/cli.py", line 9, in __init__
    getattr(self, command)()
  File "/home/usability/.local/lib/python2.7/site-packages/muselsl/cli.py", line 51, in stream
    args.interface, args.name, args.ppg, args.acc, args.gyro, args.disable_eeg)
  File "/home/usability/.local/lib/python2.7/site-packages/muselsl/stream.py", line 75, in stream
    'Muse%s' % address)
  File "/home/usability/.local/lib/python2.7/site-packages/pylsl/pylsl.py", line 193, in __init__
    c_char_p(str.encode(source_id)))
TypeError: descriptor 'encode' requires a 'str' object but received a 'unicode'

Thanks Guys :)!

oori, 2020-01-02:
`descriptor 'encode' requires a 'str' object but received a 'unicode'` says the the device name `MUSE1234` is for some reason unicode type, and not string.     Not sure why, but it doesn't seem a muselsl, nor a pylsl issue, but something in your setup.         I would start by running python3, not python2.7 (ie. `pip3 install muselsl`)

skg91, 2020-01-03:
I had the same issue. I fixed it by editing "/home/usability/.local/lib/python2.7/site-packages/pylsl/pylsl.py" line 193 c_char_p(str.encode(source_id))) --> c_char_p(unicode.encode(source_id)))

closed 2021-01-10 by jdpigeon