alexandrebarachant/muse-lsl #115: Connection Failing and Missing Samples

llamasking, 2019-12-30:
Hello, I am running on Manjaro with a cheap, fake Cambridge Silicon Radio, Ltd Bluetooth Dongle, running a patch (https://wiki.archlinux.org/index.php/Bluetooth#CSR_Dongle_0a12:0001).

When I first used the muse, I had no issue with connection or anything. It worked flawlessly from a range, however since then, I have been having issues with failing to connect and missing samples if it does connect.

oori, 2020-01-02:
Do you have a BLED112 to test?    see: https://github.com/alexandrebarachant/muse-lsl/issues/110#issuecomment-566017757

llamasking, 2020-01-02:
I saw that post, however since I have a Muse 2016 I supposed it wasn't trying to stream anything other than EEG and didn't apply to me. 

I unfortunately do not have a BLED112 and don't have any time to wait for shipping. Anything I get must be available in a store. 

I plan to get an Insignia BT dongle which has a  BCM20702A0 to test on that, too. 

My laptop, mentioned above, has a Centrino 2230 and had a similar issue where it would drop samples and disconnect at any range over a foot or two. For some reason it's spitting gatttool errors now, though. I have no idea what changed.

llamasking, 2020-01-06:
The issue was definitely the dongle. The replacement works perfectly.

closed 2020-01-06 by llamasking