alexandrebarachant/muse-lsl #117: Can not stream to desktop, problem with bluetooth adapter?

JadinTredup, 2020-01-11:
Hello, I keep trying to stream data from the Muse to my desktop but after setting root privileges for hcitool as in the common issues, I get the following error when running the "list" command. I have tried reseting ym bluetooth adapter in every known way and still have no luck.

`muselsl list
Searching for Muses, this may take up to 10 seconds...
Traceback (most recent call last):
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pexpect/pty_spawn.py", line 459, in read_nonblocking
    incoming = super(spawn, self).read_nonblocking(size)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pexpect/spawnbase.py", line 171, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pygatt/backends/gatttool/gatttool.py", line 342, in scan
    scan.expect('foooooo', timeout=timeout)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pexpect/spawnbase.py", line 341, in expect
    timeout, searchwindowsize, async_)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f2ebf4875f8>
command: /usr/bin/hcitool
args: ['/usr/bin/hcitool', '-i', 'hci0', 'lescan']
buffer (last 100 chars): b''
before (last 100 chars): b'Set scan parameters failed: Input/output error\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 1
flag_eof: True
pid: 4114
child_fd: 6
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'foooooo')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/bin/muselsl", line 11, in <module>
    load_entry_point('muselsl==2.0.2', 'console_scripts', 'muselsl')()
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/muselsl/__main__.py", line 57, in main
    cli = CLI(args.command)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/muselsl/cli.py", line 9, in __init__
    getattr(self, command)()
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/muselsl/cli.py", line 22, in list
    list_muses(args.backend, args.interface)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/muselsl/stream.py", line 32, in list_muses
    devices = adapter.scan(timeout=MUSE_SCAN_TIMEOUT)
  File "/home/jadin/Documents/Programming/TestEnvironments/BCI/venv/lib/python3.6/site-packages/pygatt/backends/gatttool/gatttool.py", line 353, in scan
    raise BLEError(message)
pygatt.exceptions.BLEError: BLE adapter requires reset after a scan as root- call adapter.reset()
`

JadinTredup, 2020-01-11:
``

gsajko, 2020-02-09:
this is problem with pygatt, so this issue should be raised there.

gsajko, 2020-02-09:
I think it's:
a) an issue with pygatt (and should be raised at their repo)
b) an issue with your BT adapter. Does it support BLE?

Teddyz, 2021-05-28:
I have a very similar output on Linux Mint. 
I am using BT v5.0 (HCI, LMP=9.x and VID=0A12 PID=0001) generic/cheap dongle on my desktop computer.
Was the original problem solved?
Is my best chance to get it to work that I buy a BLED112 dongle? (When I tried to run muselsl on Win7, it clearly said "pygatt.backends.bgapi.exceptions.BGAPIError: Unable to auto-detect BLED112 serial port")

```
minus@linuxmint:~$ sudo muselsl list
Backend was 'auto' and bluetoothctl was found, using to list muses...
Searching for Muses, this may take up to 10 seconds...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 485, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 171, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/muselsl/stream.py", line 87, in _list_muses_bluetoothctl
    scan.expect('foooooo', timeout=timeout)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 340, in expect
    return self.expect_list(compiled_pattern_list,
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f68ce2afc10>
command: /usr/bin/bluetoothctl
args: ['/usr/bin/bluetoothctl', 'scan', 'on']
buffer (last 100 chars): b''
before (last 100 chars): b'Failed to start discovery: org.bluez.Error.InProgress\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 55282
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'foooooo')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/muselsl", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/muselsl/__main__.py", line 57, in main
    cli = CLI(args.command)
  File "/usr/local/lib/python3.8/dist-packages/muselsl/cli.py", line 9, in __init__
    getattr(self, command)()
  File "/usr/local/lib/python3.8/dist-packages/muselsl/cli.py", line 32, in list
    list_muses(args.backend, args.interface)
  File "/usr/local/lib/python3.8/dist-packages/muselsl/stream.py", line 31, in list_muses
    return _list_muses_bluetoothctl(MUSE_SCAN_TIMEOUT)
  File "/usr/local/lib/python3.8/dist-packages/muselsl/stream.py", line 91, in _list_muses_bluetoothctl
    raise ValueError(msg)
ValueError: Unexpected error when scanning: Failed to start discovery: org.bluez.Error.InProgress
```