alexandrebarachant/muse-lsl #120: Muse S?

adamwolf, 2020-02-16:
I'm thinking of getting a Muse S.  Has anyone played with getting it to work with this yet?

alexandrebarachant, 2020-02-16:
Not me, but I'm curious to know if it works.

adamwolf, 2020-02-16:
I have some experience reversing BLE stuff, and a lot of experience with
BLE as a developer, so I'm pretty excited to see what's what.  I just
ordered one, but they're not shipping for a while...

RemyRamdam, 2020-03-07:
Got one yesterday, working exactly as the Muse 2 from a bluepy perspective.

iPsych, 2020-03-27:
@RemyRamdam 
Fantastique, pioneer! Thanks for the information.
Would you check how many EEG channels in Muse-S? Same four like the other bands?

Remy17, 2020-03-27:
Hello @iPsych,
As far as I can tell, they are the same
[image]

kowalej, 2020-04-05:
@Remy17 @RemyRamdam - just to confirm - were you able to steam all channels (accelerometer, gryo, PPG, EEG, telemetry, etc) without any modification of muse-lsl?

Remy17, 2020-04-06:
Sorry, I didn't notice that I had two accounts :D 
Actually, I use my own code, 99% inspired by MuseLSL but retailored to serve my purposes.
This should totally work the same with Muse LSL.

What I obtained so far is this output for the LSL streams attached to the BLE data : 
('Number of streams = ', 5)
<?xml version="1.0"?>
<info>
	<name>anonymous</name>
	<type>RAW_PPG</type>
	<channel_count>3</channel_count>
	<nominal_srate>64</nominal_srate>
	<channel_format>float32</channel_format>
	<source_id>00:55:DA:B9:09:99_PPG</source_id>
	<version>1.1000000000000001</version>
	<created_at>186968.33128078</created_at>
	<uid>13fdd4ea-7337-45f5-9044-d6ccf96817e2</uid>
	<session_id>default</session_id>
	<hostname>laptop</hostname>
	<v4address />
	<v4data_port>16574</v4data_port>
	<v4service_port>16574</v4service_port>
	<v6address />
	<v6data_port>16575</v6data_port>
	<v6service_port>16575</v6service_port>
	<desc>
		<manufacturer>Muse</manufacturer>
		<channels>
			<channel>
				<label>AMBIANT</label>
				<unit>mmHg</unit>
				<type>RAW_PPG</type>
			</channel>
			<channel>
				<label>INFRARED</label>
				<unit>mmHg</unit>
				<type>RAW_PPG</type>
			</channel>
			<channel>
				<label>RED</label>
				<unit>mmHg</unit>
				<type>RAW_PPG</type>
			</channel>
		</channels>
	</desc>
</info>

<?xml version="1.0"?>
<info>
	<name>anonymous</name>
	<type>RAW_GYRO</type>
	<channel_count>3</channel_count>
	<nominal_srate>52</nominal_srate>
	<channel_format>float32</channel_format>
	<source_id>00:55:DA:B9:09:99_GYRO</source_id>
	<version>1.1000000000000001</version>
	<created_at>186968.53343392801</created_at>
	<uid>432b0469-9c94-413b-9466-5ee4eafece50</uid>
	<session_id>default</session_id>
	<hostname>laptop</hostname>
	<v4address />
	<v4data_port>16576</v4data_port>
	<v4service_port>16576</v4service_port>
	<v6address />
	<v6data_port>16577</v6data_port>
	<v6service_port>16577</v6service_port>
	<desc>
		<manufacturer>Muse</manufacturer>
		<channels>
			<channel>
[... 437 lines cut ...]
while True :

	streams = cr.results()

	if len(streams) == 0 :
		print("Lost every stream, exiting ...")
		sys.exit()

	if len(streams) != size_streams :
		for inlet in inlets :
			del inlet
		size_streams = len(streams)
		for s in streams :
			inlets.append(StreamInlet(s))

	for inlet in inlets :
		sample, timestamp = inlet.pull_sample(timeout = [.1])
		if sample != None :
				last_timestamp = timestamp
				print("Name : " + inlet.info().name() + ", type : " + inlet.info().type() + " at " + str(timestamp) + " >> ")
				for s in sample :
					print s,
				print ""

Hope it helps ;)

Remy17, 2020-04-06:
Also, those are the outputs of gatttool : 

`gatttool -I -b 00:55:DA:B9:09:99 -t public`

```
> primary
attr handle: 0x0001, end grp handle: 0x0004 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x0005, end grp handle: 0x000b uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x000c, end grp handle: 0x0042 uuid: 0000fe8d-0000-1000-8000-00805f9b34fb

```
> char-desc 0x000c 0x0042
handle: 0x000c, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x000d, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x000e, uuid: 273e0001-4c4d-454d-96be-f03bac821358
handle: 0x000f, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0010, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0011, uuid: 273e0008-4c4d-454d-96be-f03bac821358
handle: 0x0012, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0013, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0014, uuid: 273e0009-4c4d-454d-96be-f03bac821358
handle: 0x0015, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0016, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0017, uuid: 273e000a-4c4d-454d-96be-f03bac821358
handle: 0x0018, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0019, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x001a, uuid: 273e000b-4c4d-454d-96be-f03bac821358
handle: 0x001b, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x001c, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x001d, uuid: 273e0002-4c4d-454d-96be-f03bac821358
handle: 0x001e, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x001f, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0020, uuid: 273e0003-4c4d-454d-96be-f03bac821358
handle: 0x0021, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0022, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0023, uuid: 273e0004-4c4d-454d-96be-f03bac821358
handle: 0x0024, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0025, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0026, uuid: 273e0005-4c4d-454d-96be-f03bac821358
handle: 0x0027, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0028, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0029, uuid: 273e0006-4c4d-454d-96be-f03bac821358
handle: 0x002a, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x002b, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x002c, uuid: 273e0007-4c4d-454d-96be-f03bac821358
handle: 0x002d, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x002e, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x002f, uuid: 273e000c-4c4d-454d-96be-f03bac821358
handle: 0x0030, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0031, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0032, uuid: 273e000d-4c4d-454d-96be-f03bac821358
handle: 0x0033, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0034, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0035, uuid: 273e000e-4c4d-454d-96be-f03bac821358
handle: 0x0036, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0037, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0038, uuid: 273e000f-4c4d-454d-96be-f03bac821358
handle: 0x0039, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x003a, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x003b, uuid: 273e0010-4c4d-454d-96be-f03bac821358
handle: 0x003c, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x003d, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x003e, uuid: 273e0011-4c4d-454d-96be-f03bac821358
handle: 0x003f, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0040, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0041, uuid: 273e0012-4c4d-454d-96be-f03bac821358
handle: 0x0042, uuid: 00002902-0000-1000-8000-00805f9b34fb

```

[... 50 lines cut ...]
handle: 0x0032, uuid: 273e000d-4c4d-454d-96be-f03bac821358
handle: 0x0033, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0034, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0035, uuid: 273e000e-4c4d-454d-96be-f03bac821358
handle: 0x0036, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0037, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0038, uuid: 273e000f-4c4d-454d-96be-f03bac821358
handle: 0x0039, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x003a, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x003b, uuid: 273e0010-4c4d-454d-96be-f03bac821358
handle: 0x003c, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x003d, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x003e, uuid: 273e0011-4c4d-454d-96be-f03bac821358
handle: 0x003f, uuid: 00002902-0000-1000-8000-00805f9b34fb

```

So MuseS - Muse2 is only 
```
handle: 0x0040, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0041, uuid: 273e0012-4c4d-454d-96be-f03bac821358
handle: 0x0042, uuid: 00002902-0000-1000-8000-00805f9b34fb
```

But saying that, apart from those I use in my code, I don't know what others characteristics are for.

apavlo89, 2020-09-18:
Question for people who've connected muse s. How is it the same? Are electrode column locations the same? Hasn't muse s scrapped mastoid/ear electrodes?

oori, 2020-09-18:
@apavlo89 No. same electrode positions, different material and mechanics.

ErikBjare, 2020-12-07:
FWIW, I've been able to stream EEG and PPG from the Muse S without any modifications. I'd expect ACC and GYRO to work as well, but haven't tested it.

xloem, 2021-01-06:
sorry i missed this thread when i opened #142 .  as mentioned there my notes are at https://github.com/xloem/pymuse/blob/master/muse_async.py including all gatt channels I found.  I'm new to BLE and don't really understand how the uuids relate to the handles to fill in the handles remy17 describes but they're all there; there's a thermistor and another aux channel in addition to the previously-known channels.  #143 #144 preset 0x63 streams all the channels at once which is nice.

xloem, 2021-01-07:
> But saying that, apart from those I use in my code, I don't know what others characteristics are for.
> When I tried to read the unknown characteristic, I had data coming from 0x0012, 0x0039, 0x003c, 0x003f (both Muse2 and S) and 0x0042 (just MuseS), but I don't know their meaning

0x12 is DRL_REF.  0x39-0x3f you've already described.  0x42 appears to be thermistor.

> And nothing was coming from 0x0030, 0x0033 and 0x0036 (but this could come from a disabled parameter somewhere).

I have magnetometer, pressure, ultraviolet for these.  I haven't seen them output anything either.

xloem, 2022-04-05:
I just recently noticed that the muse S appears to advertise its serial port over the usb cable attached to it. It looks like the muse S can be used without bluetooth at all, exchanging packets over its usb cable. I didn't know this. It would be good to eventually support direct serial access.

guz6, 2023-02-25:
When I connect to MuseS, I can not get the extra electrode signal, which is Right AUX channel. Does anyone know how to fix this issue? How to modify the code? Thank you very much.