alexandrebarachant/muse-lsl #123: Cannot simultaneously recording EEG and PPG

iPsych, 2020-03-26:
Still same issue. 
[]https://github.com/alexandrebarachant/muse-lsl/issues/94

The PPG signal cannot be recorded simultaneously with EEG.
Current environment is MacOS High-Sierra, Python3.7, and below errors are happens.
As jdpigeon told, disable EEG works, but than no use for real application...

```
from muselsl import stream, list_muses
muses = list_muses()
Searching for Muses, this may take up to 10 seconds...
Found device Muse-1913, MAC Address 00:55:DA:B7:19:13
stream(muses[0]['address'], ppg_enabled=True)
Connecting to Muse: 00:55:DA:B7:19:13...
Connected.
Streaming EEG PPG...
```

```
muselsl record --type PPG
Looking for a PPG stream...
Can't find PPG stream.
```

oori, 2020-04-22:
See: https://github.com/alexandrebarachant/muse-lsl/issues/110

iPsych, 2020-04-24:
@oori 
The same issue happens as below.

Looking for a PPG stream...
Can't find PPG stream.

PPG stream only works with --disable-eeg option...

iPsych, 2020-04-24:
@oori 
Also tested all possible pair (i.e. gyro +PPG, gyro +ACC, ACC+EEG etc...),
and found only one stream works. (pylsl 1.13.1, pygatt 4.0.5 with BLED112 in Mac OS High-Sierra).

spanezz, 2023-03-15:
I observe the same: only the last opened lsl outlet gets sent.

I tried to reproduce a multi-outlet sending with this script, and indeed only the last one will appear in lsl outlet resolution:
```py
#!/usr/bin/python3

import time

import pylsl

def main():
    info1 = pylsl.StreamInfo(
            "POS", "Test", 1, pylsl.IRREGULAR_RATE, 'float32', 'myuid34234')
    info2 = pylsl.StreamInfo(
            "NEG", "Test", 1, pylsl.IRREGULAR_RATE, "int16", 'myuid43432')

    # next make an outlet
    outlet1 = pylsl.StreamOutlet(info1)
    outlet2 = pylsl.StreamOutlet(info2)

    seq: int = 0
    while True:
        outlet1.push_sample([time.time()])
        if seq % 3 == 0:
            outlet2.push_sample([seq % 65536])
        seq += 1
        time.sleep(0.01)

if __name__ == '__main__':
    main()
```

spanezz, 2023-03-15:
It looks like a firewall issue: if I stop the firewall, all streams show up in discovery