alexandrebarachant/muse-lsl #124: Streaming using the Muse 2 with Right Aux attached shows flat line for right aux.

MalusGreen, 2020-03-26:
Streaming with the muse 1 results in a perfectly good connection with our aux electrode. However, using a muse 2, though a connection is good/stable, and though we are able to retrieve the data using our own legacy scripts, muselsl seems to not be streaming any relevant data for the right aux. Resulting in a flat line.

Is this a result of the GATT ATTR for Right Aux being pointed to the wrong address?

stepheninx, 2020-06-25:
It looks as though self.select_preset was not being called during muse.start(). Since no preset command was sent, the headset would only stream the default data channels.

I have opened a pull request here with a fix: <https://github.com/alexandrebarachant/muse-lsl/pull/128>

For fun, here is a picture of the AUXR electrode being used to measure my heartbeat as a test.
[image]

victorgutemberg, 2023-10-28:
@MalusGreen I'm trying to find where to buy these aux USB electrodes. Could you please share where you found them?