alexandrebarachant/muse-lsl #127: Windows BlueMuse Streaming Wrong Marker Timestamp

simonguozirui, 2020-06-18:
Hi, this is a problem I have stuck myself on for a couple of days and found a very subtle fix. 
I hope this can help others that have also been stuck with similar issues.

OS `Windows 10`
Python Runtime `2.7.18` (configured in a `conda` environment)
No Bluetooth Dongle 
Muse model: Muse 1 2016

I use BlueMuse to stream data from my Muse via LSL to my computer. I run some ERP experiments from the notebooks. However, the marker would not label correctly. Only the first entry was labeled with a marker, causing the data useless for analysis. After lots of debugging, I realized the timestamp in my recorded data and the marker timestamp are order of magnitudes different. The logic of recording code was not wrong after thinking about it, so there must be some issues within the input.

Thus I looked into BlueMuse and I found one of the settings ⚙️, in the `Primary TimeStamp Format` > `Main LSL Stream TimeStamp`. It is defaulted to be `BlueMuse High Accuracy (Unix Epoch Seconds UTC -0)`. You need to change it to `native LSL Local Clock - Via Bridge (System Uptime Seconds)`. Do this every time you start BlueMuse!

Try this if you have issues with marker being incorrectly registered! Feel free to add this to the documentation too!

jdpigeon, 2020-06-19:
Ahh, sorry you ran into this. I remember @kowalej had to do a lot of work a few years ago working around the fact that windows was creating timestamps of different format than linux and Mac machines.

Just to be clear, are the timestamps generated in the notebooks in unix epoch seconds or system uptime? I think we should make sure they're in unix epoch for all platforms.

buyuk-dev, 2021-08-15:
seems related to https://github.com/alexandrebarachant/muse-lsl/issues/130

closed 2023-04-24 by simonguozirui