alexandrebarachant/muse-lsl #133: Native bluetooth on macOS? It seems uvicMUSE was able to do it

jacksonmgoode, 2020-10-07:
[uvicMUSE](https://github.com/bardiabarabadi/uvicMUSE) was a package I stumbled across while trying to interpret the Muse (2016) sensor streams for a sonification project in Puredata. Ideally, muse-lsl would be the fit, for a simple cli and far more documentation - but it requires a bluetooth dongle for Mac?

In trying uvicMUSE, it looks like there is a stable connection to the Muse headband over the native bluetooth device on my Macbook. Wondering if there is anyway to incorporate their methods into muse-lsl? Or is there another reason?

jacksonmgoode, 2020-10-11:
It seems they use [bleak](https://github.com/hbldh/bleak) to search for BLE devices rather than pygatt in muse-lsl's case.

buyuk-dev, 2021-08-15:
There is an attempt to add bleak backend to muselsl. You can take a look at it in one of this PRs: https://github.com/alexandrebarachant/muse-lsl/pull/148  - this is the first attempt which I overlooked while implementing my own version here: https://github.com/alexandrebarachant/muse-lsl/pull/166.

Feel free to try either of those, I've tested my change with muse 2, MacBook Pro 2015 and the basic functionality seems to work (muselsl list, muselsl stream, muselsl view).