alexandrebarachant/muse-lsl #135: RuntimeError: in pylsl.py. The library liblsl32.so was not found.

AbhisarAnand, 2020-10-11:
Hello,

I have been trying to run this code but whatever I do, I run into this RuntimeError. I have the file in the same folder and I even changed the permissions so to 777 but it keeps failing.

Here is the error message we keep getting:
`Traceback (most recent call last):
  File "muse-lsl.py", line 3, in <module>
    from pylsl import StreamInfo, StreamOutlet, local_clock
  File "/home/pi/.virtualenvs/py3cv4/lib/python3.7/site-packages/pylsl/__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "/home/pi/.virtualenvs/py3cv4/lib/python3.7/site-packages/pylsl/pylsl.py", line 1197, in <module>
    raise RuntimeError("library " + libname + " was not found - make sure "
RuntimeError: library liblsl32.so was not found - make sure that it is on the search path (e.g., in the same folder as pylsl.py).
`
Can someone help us as soon as possible.

Thank you,
Abhisar Anand

oori, 2020-10-26:
Maybe related to this: https://github.com/alexandrebarachant/muse-lsl/issues/137#issuecomment-716782200

plnnr, 2021-02-15:
I added a potentially relevant comment [here](https://github.com/alexandrebarachant/muse-lsl/issues/137#issuecomment-779016836).